monkeymademe/CamUI #10: Add ColorGains Setting

monkeymademe, 2024-02-05:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

monkeymademe, 2024-02-16:
ColourGains might need more code than I thought. Default values are as always min, max and default but they are a single float number. Picamera2 (or rather libcamera at the end) expects 2 floating point numbers for Red and Blue.

So a decision needs to be made. Do I collect the values for the ColourGains from the feed and use them as default. Zero the floating point numbers or not apply any ColourGains and allow the user to set them (basically exclude them from the default values)

TimPGamer, 2024-04-20:
yo that would be nice for people using night vision cameras, to correct some of the redness when using. The infrared, that makes its way into the camera makes the image quite a bit red. I would appreciate it if you could implement it.