monkeymademe/CamUI #15: create_video_configuration() got an unexpected keyword argument 'sensor'

leighghunt, 2024-02-08:
**Describe the bug**
On first clone and run, get the following error:

```
pi@sediment-pi-zero-w-v1-f:~/dev/picamera2-WebUI-Lite$ python app.py 
[0:03:10.910392876] [1483]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:03:11.231004734] [1768]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media0
[0:03:11.236952283] [1768]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:03:11.331264217] [1483]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-XBGR8888 (1) 1536x864-SBGGR10_CSI2P
[0:03:11.340315720] [1768]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
[0:03:11.495844626] [1483]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-XBGR8888 (1) 2304x1296-SBGGR10_CSI2P
[0:03:11.503643613] [1768]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
[0:03:11.606357925] [1483]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-XBGR8888 (1) 4608x2592-SBGGR10_CSI2P
[0:03:11.614973596] [1768]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected unicam format: 4608x2592-pBAA
Traceback (most recent call last):
  File "/home/pi/dev/picamera2-WebUI-Lite/app.py", line 23, in <module>
    video_config = picam2.create_video_configuration(sensor={'output_size': mode['size'], 'bit_depth': mode['bit_depth']})
TypeError: create_video_configuration() got an unexpected keyword argument 'sensor'
```

**To Reproduce**
Steps to reproduce the behavior:
1. `python app.py`

**Desktop (please complete the following information):**
 - OS: Raspbian GNU/Linux 11 (bullseye)

**Additional context**
Assuming this is a picamera2 version issue - `sensor` argument seems to have been added in https://github.com/raspberrypi/picamera2/commit/fa69488488662d99a1629bfda6367033baa57471. Raising bug to aid anyone else experiencing this issue - hoping to close with solution shortly.

leighghunt, 2024-02-08:
Looks like I'm running a version from just before that argument was added.

```
libcamera-hello --version
libcamera-apps build: 7e4d3d71867f 17-07-2023 (07:34:37)
libcamera build: v0.0.5+83-bde9b04f
```

leighghunt, 2024-02-08:
Got it working by checking out version 6de5578355119c5ca7b476e33ab0b88c1478faae. I suspect upgrading Raspbian to Bookworm would probably update picamera. I'll try this soon.

Getting about 1 frame every 1-2 seconds on a Pi Zero W, with a 12 second delay, when running over a 4G/LTE cellular modem over Tailscale VPN.

[image]

monkeymademe, 2024-02-08:
yeah this sensor argument was added in yesterday did not test it on bullseye... Bookworm is best to be honest everything is up to date there. Bullseye is considered legacy...

But thanks for the note I will change the readme to be a little more explicit on the versions

monkeymademe, 2024-02-08:
Going to close this for now due to how legacy bullseye is

closed 2024-02-08 by monkeymademe