monkeymademe/CamUI #22: Can't connect to web interface + instructions errors

Oceanique, 2024-02-28:
**Describe the bug**
Can't connect to web interface after trying to connect to "http://Your IP:8080/" with my laptop while being connected to the Pi via SSH (to be clear, I replaced Your IP by the raspberry pi IP)

**To Reproduce**
Enter http://Your IP:8080/ and it fails to connect

**Expected behavior**
Expecting to connect to the web interface

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Linux Mint last version
 - Firefox
 - Mint version 

**Additional context**
I've struggled following the instructions in the README.txt 

First of all, the "flask" link leads to a tutorial saying to create a virtual environment to install it, so after following the instructions, it couldn't find it because it was installed in another venv. To install flask, I ran "sudo apt install python3-flask".

The instruction to git clone the repository asks for a username and password "git clone https://github.com/your-username/picamera2-WebUI-Lite.git", I had to git clone the URL I was on "git clone https://github.com/monkeymademe/picamera2-WebUI-Lite" to make it work.

After that, the 3 and 4 steps worked fine. 

But when trying to connect to the web interface through my laptop browser (the RPi 4B still connected to my laptop through SSH), Firefox or Chromium say "connection failed" like there's nothing.

Running netstat -tln results in : 
```

Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State      
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN     
tcp6       0      0 :::22                   :::*                    LISTEN 
```
While searching on the web, someone encountering appx the same issue was told the RPi maybe couldn't choose between Wifi or Ethernet, but I don't understand anything to fix this if this is the issue.

monkeymademe, 2024-02-28:
Hey there sorry you are having issues... to be perfectly honest I also don't like the virtual env thing either... If you try and run the pip install without the venv thing it will fail to install but also says there is a switch you can apply to the command to break the way its installed '--break-system-packages' basically it forcibly installs flask to the system. This is not recommended and you should learn how the venv works but I get you might be frustrated starting out so this is the cheat... 

After that I need to know more about what happens with step 4 can you give me the last few lines of the terminal output when you run 'python app.py'

When I run it I get the following - 
[image]

I am also not sure about Linux mint I have not tested on that OS. Only Raspberry Pi OS Bookworm so you might have issues with the picamera2 not being installed and also libcamera? For the sake of getting it working quickly can you switch to Raspberry Pi OS bookworm?

Your netstat is telling me your server is not running so I need more information to point you in the right direction.

monkeymademe, 2024-02-28:
Oh I should say if linux mint is missing a dependancy like picamera2 or libcamera or the camera is not found the server will not run. Its not great I know...

Oceanique, 2024-02-28:
Thanks for your fast reply :).

My RPi 4B is running on Bookworm Lite 64 bits and if I install Flask with "apt install python3-flask" without any env, I get the same output you showed on your screenshot, but still, I can't connect to the web interface :/

Oceanique, 2024-02-29:
`nico@pi-webcam:~/picamera2-WebUI-Lite $ python app.py
[0:08:41.074562063] [2040]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[0:08:41.103362978] [2043]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:08:41.105688137] [2043]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
[0:08:41.105768488] [2043]  INFO RPI pipeline_base.cpp:1144 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:08:41.108634622] [2040]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[0:08:41.135911220] [2046]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:08:41.138051826] [2046]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
[0:08:41.138122714] [2046]  INFO RPI pipeline_base.cpp:1144 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
{'controls': {'LensPosition': 1.0, 'ExposureValue': 0.0, 'ExposureTime': 220417486, 'AwbMode': 0, 'AfSpeed': 0, 'AfRange': 0, 'AeMeteringMode': 0, 'AeFlickerPeriod': 1000000, 'AeFlickerMode': 0, 'AeExposureMode': 0, 'AeEnable': True, 'AwbEnable': True, 'AeConstraintMode': 0, 'Brightness': 0.0, 'Contrast': 1.0, 'Saturation': 1.0, 'Sharpness': 1.0, 'AfMode': 0, 'ScalerCrop': [0, 0, 4608, 2592]}, 'rotation': {'hflip': 0, 'vflip': 0}, 'sensor-mode': 2, 'capture-settings': {'Resize': False, 'makeRaw': True, 'Resolution': 0, 'available-resolutions': [[4608, 2592], [2304, 1296], [1920, 1080], [1280, 720], [640, 360]]}}
{'Resize': False, 'makeRaw': True, 'Resolution': 0, 'available-resolutions': [[4608, 2592], [2304, 1296], [1920, 1080], [1280, 720], [640, 360]]}
[4608, 2592]
[0:08:41.145044765] [2040]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 1332x990-SBGGR10_CSI2P
[0:08:41.145491815] [2046]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 1332x990-SBGGR10_1X10 - Selected unicam format: 1332x990-pBAA
[0:08:41.157969415] [2040]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 2028x1080-SBGGR12_CSI2P
[0:08:41.158385225] [2046]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1080-SBGGR12_1X12 - Selected unicam format: 2028x1080-pBCC
[0:08:41.171873405] [2040]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 2028x1520-SBGGR12_CSI2P
[0:08:41.172290845] [2046]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[0:08:41.188120166] [2040]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 4056x3040-SBGGR12_CSI2P
[0:08:41.189431984] [2046]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
{'use_case': 'video', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'Rec709'>, 'buffer_count': 6, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (4608, 2592)}, 'lores': None, 'raw': {'format': 'SRGGB12_CSI2P', 'size': (4608, 2592)}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (33333, 33333)}, 'sensor': {'output_size': (2028, 1520), 'bit_depth': 12}, 'display': 'main', 'encode': 'main'}
{'Model': 'imx477', 'UnitCellSize': (1550, 1550), 'ColorFilterArrangement': 0, 'Location': 2, 'Rotation': 180, 'PixelArraySize': (4056, 3040), 'PixelArrayActiveAreas': [(8, 16, 4056, 3040)], 'ScalerCropMaximum': (0, 0, 4056, 3040), 'SystemDevices': (20750, 20751, 20737, 20738, 20739), 'SensorSensitivity': 1.0}
INFO:picamera2.picamera2:Camera configuration has been adjusted!
[0:08:41.241432414] [2040]  INFO Camera camera.cpp:1183 configuring streams: (0) 4608x2592-XBGR8888 (1) 2028x1520-SBGGR12_CSI2P
[0:08:41.241798317] [2046]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
INFO:picamera2.picamera2:Configuration successful!
INFO:picamera2.picamera2:Camera started
 * Serving Flask app 'app'
 * Debug mode: off
INFO:werkzeug:WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on http://127.0.0.1:8080
 * Running on http://192.168.1.49:8080
INFO:werkzeug:Press CTRL+C to quit

`

I've tried with Bookworm with desktop, the script runs well but I can't access to the web interface still.. :/

monkeymademe, 2024-02-29:
ok so with the python app.py still running you open another terminal window and run the netstat command and it still shows you what you showed me before...

The server is not persistent if you close the window it will turn off... 

If your wifi network 192.168.1.* or is that wired connection to the pi?

On the bookworm with desktop what happens when you goto the http://192.168.1.49:8080/ url when your running the server?

Oceanique, 2024-02-29:
I'm using VNC server because I don't have a monitor. The script starts as expected, but the VNC freezes and the terminal and firefox close, so I can't know if the web interface is working while being on the RPi desktop...

monkeymademe, 2024-02-29:
2 things... its possible that the pi is rebooting cause its not getting enough power to run the camera and the pi at the same time. I had that exact same issue. Its due to a poor cable or USB connector (my case was a bad USB plug). If VNC is dropping it might be that... maybe

2nd you don't need desktop to run this you could SSH into the pi and run it from terminal.

I would recommend running the server in 'screen' so that the instance remains running even if you close the terminal window

To install screen 'sudo apt install screen'

The type 'screen python app.py' then you can close the terminal window and the server will still be running. If you want to get back to the server you can type 'screen -x' and you will return the log output

Also Tmux is an alternative to screen but i have never personally used it but there is an awesome video explaining how it works - https://www.youtube.com/watch?v=nTqu6w2wc68

Oceanique, 2024-02-29:
Thanks, I forgot about that, I just ordered the right charger and I'll try it tomorrow

Oceanique, 2024-03-03:
Got it working ! Even with the right charger, I couldn't connect to the Pi IP address, so I searched online for other projects. Then I found an old tutorial to live stream on any devices, which is exactly what you've done but with settings.

This old tutorial from Github proposed to install those dependencies, I installed a lot of them by using "--break-system-packages" : 

`sudo apt-get update
sudo apt-get upgrade

sudo apt-get install libatlas-base-dev
sudo apt-get install libjasper-dev
sudo apt-get install libqtgui4
sudo apt-get install libqt4-test
sudo apt-get install libhdf5-dev

sudo pip3 install flask
sudo pip3 install numpy
sudo pip3 install opencv-contrib-python
sudo pip3 install imutils
sudo pip3 install opencv-python`

Then, I tried the project, it didn't work, but before shutting down the Pi, I wanted to know if I was able to connect with your python script on the Pi itself, it worked ! So I shut down the VNC, connected my laptop to the address and voilà ! It's working like a charm.

Right now, I want to know which dependencies made it work, because it has to have a link with one or a few of those dependencies. I will try later with another SD card and install one by one the dependencies so I know which one make it work. But the only one that's interesting is obviously Flask, which I installed without a venv with "--break-system-packages" using pip3.

Thank you, and I didn't know you created a phone interface too !, it's really useful to capture pictures with the phone.

monkeymademe, 2024-03-04:
Should just be flask to be honest at least that is from the pi OS not mint ... So if picamera2 has other dependencies I am not aware of that might be an issue. I am going to close this issue now cause you are not up and working

closed 2024-03-04 by monkeymademe

Oceanique, 2024-03-20:
Hello,

Latest RPiOS Bookworm img is fully updated, no need to install Flask or Picamera2, they are already installed :).

Just had to clone the git, cd the directory and run the script, it works like a charm :)