monkeymademe/CamUI #25: RuntimeError: Failed to start camera: Cannot allocate memory

Exper1mental, 2024-03-04:
**Describe the bug**
When I run `python app.py` I am given the error: `RuntimeError: Failed to start camera: Cannot allocate memory`.

Here is the full output:
```bash
python app.py 
[1:11:58.421579735] [5828]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[1:11:58.459215937] [5841]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:11:58.462816783] [5841]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media1
[1:11:58.463588051] [5841]  INFO RPI pipeline_base.cpp:1144 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[1:11:58.468412550] [5828]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[1:11:58.502212389] [5844]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:11:58.506007159] [5844]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media1
[1:11:58.506750743] [5844]  INFO RPI pipeline_base.cpp:1144 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
{'controls': {'LensPosition': 1.0, 'ExposureValue': 0.0, 'ExposureTime': 220417486, 'AwbMode': 0, 'AfSpeed': 0, 'AfRange': 0, 'AeMeteringMode': 0, 'AeFlickerPeriod': 1000000, 'AeFlickerMode': 0, 'AeExposureMode': 0, 'AeEnable': True, 'AwbEnable': True, 'AeConstraintMode': 0, 'Brightness': 0.0, 'Contrast': 1.0, 'Saturation': 1.0, 'Sharpness': 1.0, 'AfMode': 0, 'ScalerCrop': [0, 0, 4608, 2592]}, 'rotation': {'hflip': 0, 'vflip': 0}, 'sensor-mode': 2, 'capture-settings': {'Resize': False, 'makeRaw': False, 'Resolution': 0, 'available-resolutions': [[4608, 2592], [2304, 1296], [1920, 1080], [1280, 720], [640, 360]]}}
{'Resize': False, 'makeRaw': False, 'Resolution': 0, 'available-resolutions': [[4608, 2592], [2304, 1296], [1920, 1080], [1280, 720], [640, 360]]}
[4608, 2592]
[1:11:58.517771279] [5828]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 1536x864-SBGGR10_CSI2P
[1:11:58.518357883] [5844]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
[1:11:58.534220510] [5828]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 2304x1296-SBGGR10_CSI2P
[1:11:58.534733818] [5844]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
[1:11:58.554731989] [5828]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 4608x2592-SBGGR10_CSI2P
[1:11:58.555365740] [5844]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected unicam format: 4608x2592-pBAA
{'use_case': 'video', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'Rec709'>, 'buffer_count': 6, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (4608, 2592)}, 'lores': None, 'raw': {'format': 'SRGGB10_CSI2P', 'size': (4608, 2592)}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (33333, 33333)}, 'sensor': {'output_size': (4608, 2592), 'bit_depth': 10}, 'display': 'main', 'encode': 'main'}
{'Model': 'imx708_wide', 'UnitCellSize': (1400, 1400), 'ColorFilterArrangement': 0, 'Location': 2, 'Rotation': 180, 'PixelArraySize': (4608, 2592), 'PixelArrayActiveAreas': [(16, 24, 4608, 2592)], 'ScalerCropMaximum': (0, 0, 4608, 2592), 'SystemDevices': (20750, 20751, 20738, 20739, 20740), 'SensorSensitivity': 1.0}
INFO:picamera2.picamera2:Camera configuration has been adjusted!
[1:11:58.681543280] [5828]  INFO Camera camera.cpp:1183 configuring streams: (0) 4608x2592-XBGR8888 (1) 4608x2592-SBGGR10_CSI2P
[1:11:58.682449214] [5844]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected unicam format: 4608x2592-pBAA
INFO:picamera2.picamera2:Configuration successful!
[1:11:59.371621240] [5844] ERROR V4L2 v4l2_videodevice.cpp:1241 /dev/video0[16:cap]: Unable to request 2 buffers: Cannot allocate memory
[1:11:59.372477786] [5844] ERROR RPI pipeline_base.cpp:663 Failed to allocate buffers
Traceback (most recent call last):
  File "/home/admin/.local/share/Trash/files/picamera2-WebUI-Lite-main-0.0.4/app.py", line 413, in <module>
    start_camera_stream()
  File "/home/admin/.local/share/Trash/files/picamera2-WebUI-Lite-main-0.0.4/app.py", line 283, in start_camera_stream
    picam2.start_recording(JpegEncoder(), FileOutput(output))
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1761, in start_recording
    self.start()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1163, in start
    self.start_()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1131, in start_
    self.camera.start(controls)
RuntimeError: Failed to start camera: Cannot allocate memory
INFO:picamera2.picamera2:Camera closed successfully.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Download release v0.0.4 and extract to folder
2. Create new venv using `python -m venv --system-site-package venv` (to access system packages like picamera2)
3. cd into folder while in venv and run `python app.py`
4. See error

**Expected behavior**
The script start up without error.

**Desktop (please complete the following information):**
 - Pi: Raspberry Pi 4B (8GB RAM)
 - OS: Raspberry Pi OS 64-bit (Bookworm)
 - Browser: Chromium 122.0.6261.89

**Additional context**
For reference, here are the packages installed / accessible through the venv:
```bash
python -m pip list
Package                            Version
---------------------------------- ----------
arandr                             0.1.11
asgiref                            3.6.0
astroid                            2.14.2
[... 218 lines cut ...]
types-ttkthemes                    3.2
types-typed-ast                    1.5
types-tzlocal                      4.2
types-ujson                        5.5
types-urllib3                      1.26
types-vobject                      0.9
types-waitress                     2.1
types-whatthepatch                 1.0
types-xmltodict                    0.13
types-xxhash                       3.0
types-zxcvbn                       4.4
typing_extensions                  4.4.0
urllib3                            1.26.12
v4l2-python3                       0.3.3
wadllib                            1.3.6
webencodings                       0.5.1
Werkzeug                           2.2.2
wheel                              0.38.4
wrapt                              1.14.1
zipp                               1.0.0
```

What should I do to fix this?

Thanks!

monkeymademe, 2024-03-04:
Which Version of pi camera is that? V3 or the Araducam? Could be a memory allocation issue but I am not sure. This system has not been tested on the Araducam (I don't have one)

Exper1mental, 2024-03-05:
Hmm interesting... I'm using a Pi camera module 3 though. ¯\_(ツ)_/¯

monkeymademe, 2024-03-05:
what happens when you type libcamera-hello can you paste the output (do that before going into the venv) 

Also Raspberry Pi OS 64bit. --- lite or desktop ... trying to narrow down the issue

Exper1mental, 2024-03-05:
Running `libcamera-hello` out of venv displayed the popup preview window as expected with a live view from the camera and gives the following output:
```bash
libcamera-hello
[6:52:16.986787243] [7603]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[6:52:17.033919132] [7606]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[6:52:17.036115052] [7606]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media1 and ISP device /dev/media2
[6:52:17.036193327] [7606]  INFO RPI pipeline_base.cpp:1144 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Made X/EGL preview window
Mode selection for 2304:1296:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 3400
    SRGGB10_CSI2P,2304x1296/0 - Score: 1000
    SRGGB10_CSI2P,4608x2592/0 - Score: 1900
Stream configuration adjusted
[6:52:17.510516614] [7603]  INFO Camera camera.cpp:1183 configuring streams: (0) 2304x1296-YUV420 (1) 2304x1296-SBGGR10_CSI2P
[6:52:17.512707497] [7606]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
```

I'm using Raspberry Pi OS with Desktop 64-bit (Bookworm / Debian 12)

Exper1mental, 2024-03-05:
I've submitted an issue on the Picamera2 Github repo. Seems like this is a bug in the library?
https://github.com/raspberrypi/picamera2/issues/972

monkeymademe, 2024-03-05:
Going to abuse this to take the install notes

This what I did to get this working on 32bit Lite Pi OS

- sudo apt-get update
- sudo apt-get upgrade
- libcamera-hello
- sudo apt install -y python3-picamera2
- sudo apt-get install git
- git clone https://github.com/monkeymademe/picamera2-WebUI-Lite
- cd picamera2-WebUI-Lite/
- python -m venv --system-site-package venv
- source ./venv/bin/activate
- pip install flask
- python app.py

This was working for me... I am waiting on a SD card for 64bit

monkeymademe, 2024-03-05:
I had forgotten you use v0.0.4 and yeah its still functioning so I am at a loss... I did have one issue with the 64bit version but when I reseated the cable it was fine.

Do you want to clean install the OS and try the above steps

Exper1mental, 2024-03-06:
I initially was able to get this working by setting `buffer_count=5` for `picam2.create_video_configuration` (the default is `6`).

Upon further investigation, Visual Studio Code uses a small (~30 MB) amount of CMA memory and this was sufficient to cause the error. Closing VSC and running from Terminal allowed me to push the buffer count back to `6`

Closing since this isn't a bug.

## Some info for anyone who has this issue down the road:
`grep Cma /proc/meminfo` tells how much CMA memory is currently available and in-use.

`dtoverlay -h vc4-fkms-v3d` tells you what values your Pi will accept for defining CMA memory allocation.

Per page 58 (section 8.3) the [Picamera2 manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf), you can then edit the `/boot/firmware/config.txt` (previously `/boot/config.txt`) file to set CMA memory.

closed 2024-03-06 by Exper1mental

pstimpel, 2024-03-12:
Just in case it helps ... 

Raspian bookworm 32bit lite, Raspi 3B. CMA memory config changes did not help, but I raised it to 320 and confirmed the settings are in use. What helped at the end: adding buffer_count=4 (!) to the statement creating the video config.

brbergami, 2024-05-20:
I'm having a simillar issue. I guess mine is an Arducam (v1.3? It's `ov5647` at least), can't tell exactly (info command welcomed).

I'm running DietPi 9.4.2 
The error is the following:
```
root@DietPi:~/picamera2-WebUI-Lite# python3 app.py
[0:12:10.852686000] [2347]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[0:12:11.081730000] [2371]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:12:11.103053000] [2371]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media2 and ISP device /dev/media0
[0:12:11.104807000] [2371]  INFO RPI pipeline_base.cpp:1102 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:12:11.131601000] [2347]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[0:12:11.329701000] [2374]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:12:11.350595000] [2374]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media2 and ISP device /dev/media0
[0:12:11.352322000] [2374]  INFO RPI pipeline_base.cpp:1102 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'

Camera Config:
{'controls': {'LensPosition': 1.0, 'ExposureValue': 0.0, 'ExposureTime': 220417486, 'AwbMode': 0, 'AfSpeed': 0, 'AfRange': 0, 'AeMeteringMode': 0, 'AeFlickerPeriod': 1000000, 'AeFlickerMode': 0, 'AeExposureMode': 0, 'AeEnable': True, 'AwbEnable': True, 'AeConstraintMode': 0, 'Brightness': 0.0, 'Contrast': 1.0, 'Saturation': 1.0, 'Sharpness': 1.0, 'AfMode': 0, 'ScalerCrop': [0, 0, 4608, 2592]}, 'rotation': {'hflip': 0, 'vflip': 0}, 'sensor-mode': 0, 'capture-settings': {'Resize': False, 'makeRaw': False, 'Resolution': 0, 'available-resolutions': [[4608, 2592], [2304, 1296], [1920, 1080], [1280, 720], [640, 360]]}}

Camera Settings:
{'Resize': False, 'makeRaw': False, 'Resolution': 0, 'available-resolutions': [[4608, 2592], [2304, 1296], [1920, 1080], [1280, 720], [640, 360]]}

Camera Set Resolution:
[4608, 2592]

[0:12:11.439154000] [2347]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 640x480-SGBRG10_CSI2P
[0:12:11.442684000] [2374]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
[0:12:11.518807000] [2347]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 1296x972-SGBRG10_CSI2P
[0:12:11.523524000] [2374]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
[0:12:11.621510000] [2347]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 1920x1080-SGBRG10_CSI2P
[0:12:11.624426000] [2374]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA
[0:12:11.809884000] [2347]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 2592x1944-SGBRG10_CSI2P
[0:12:11.822447000] [2374]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 2592x1944-SGBRG10_1X10 - Selected unicam format: 2592x1944-pGAA

Video Config:
{'use_case': 'video', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'Rec709'>, 'buffer_count': 6, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (4608, 2592)}, 'lores': None, 'raw': {'format': 'SGBRG10_CSI2P', 'size': (4608, 2592)}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (33333, 33333)}, 'sensor': {'output_size': (640, 480), 'bit_depth': 10}, 'display': 'main', 'encode': 'main'}

Picamera2 Camera Properties:
{'Model': 'ov5647', 'UnitCellSize': (1400, 1400), 'ColorFilterArrangement': 2, 'Location': 2, 'Rotation': 0, 'PixelArraySize': (2592, 1944), 'PixelArrayActiveAreas': [(16, 6, 2592, 1944)], 'ScalerCropMaximum': (0, 0, 2592, 1944), 'SystemDevices': (20749, 20737, 20738, 20739), 'SensorSensitivity': 1.0}

INFO:picamera2.picamera2:Camera configuration has been adjusted!
[0:12:12.163953000] [2347]  INFO Camera camera.cpp:1183 configuring streams: (0) 4608x2592-XBGR8888 (1) 640x480-SGBRG10_CSI2P
[0:12:12.166702000] [2374]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
INFO:picamera2.picamera2:Configuration successful!
Traceback (most recent call last):
  File "/root/picamera2-WebUI-Lite/app.py", line 508, in <module>
    start_camera_stream()
  File "/root/picamera2-WebUI-Lite/app.py", line 304, in start_camera_stream
    picam2.configure(video_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1114, in configure
    self.configure_(camera_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1097, in configure_
    self.allocator.allocate(libcamera_config, camera_config.get("use_case"))
  File "/usr/lib/python3/dist-packages/picamera2/allocators/dmaallocator.py", line 43, in allocate
    fd = self.dmaHeap.alloc(f"picamera2-{i}", stream_config.frame_size)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/dma_heap.py", line 98, in alloc
[... cut ...]
#89 (30.01 fps) exp 33222.00 ag 7.88 dg 1.02
#90 (30.02 fps) exp 33222.00 ag 7.88 dg 1.02
#91 (30.02 fps) exp 33222.00 ag 7.88 dg 1.02
#92 (30.01 fps) exp 33222.00 ag 7.88 dg 1.02
#93 (30.01 fps) exp 33222.00 ag 7.88 dg 1.02
#94 (30.02 fps) exp 33222.00 ag 7.88 dg 1.02
#95 (30.01 fps) exp 33222.00 ag 7.88 dg 1.02
^C
```

I'm running this using python3, didn't digged too much in the repo code to tell if that's a difference, but as I found this issue opened, discarded that to be the reason. Hope the info helps.