monkeymademe/CamUI #29: Make a link to the raw video feed for ease of use with OBS

monkeymademe, 2024-03-13:
Make a link to the raw video feed for ease of use with OBS.

Top menu item - 'Raw video feed'?

geerlingguy, 2024-06-08:
Maybe since this is more an advanced feature, it could be on the Camera Info page, and be able to be easily selected for copy + paste.

I would like to try using the feed with VLC (for streaming to another computer), or maybe even with other apps on the Pi like Octoprint, since the Pi 5 integration for the Pi camera streaming isn't working yet.

monkeymademe, 2024-07-29:
Added the links to copy and its been tested working with VLC, OBS and Octoprint... Enjoy!

[image]

closed 2024-07-29 by monkeymademe

geerlingguy, 2024-07-30:
Beautiful! The only other suggestion - if you could make the links click-to-copy, like: `user-select-all`, that would make copying them out even easier (though just having the link means you can click it, then Command+L (on Mac, at least) and copy from the URL field).

monkeymademe, 2024-07-30:
No worries

[image]