monkeymademe/CamUI #31: Implement snapshot

gatonero, 2024-03-15:
**Is your feature request related to a problem? Please describe.**
In OctoPrint Stream-URL works perfectly with http://<ip-addr>8080/video_feed. But this address doesn't work for Snapshot-URL

**Describe the solution you'd like**
I would like to have an URL like  http://<ip-addr>8080/snapshot or similar.

**Additional context**
[image]

monkeymademe, 2024-03-16:
Not sure what snapshot is expecting? a Still image or another video feed (not used octoprint)

gatonero, 2024-03-16:
It should be still image.

monkeymademe, 2024-03-16:
I'll see what I can do. Since I don't have octoprint anywhere can I get you to test it when its ready?

gatonero, 2024-03-16:
Yes of corse! I have a running instance with this application. Right now I'm printing with it. :-)
[image]

gatonero, 2024-03-16:
Here is explained what OctoPrint is doing [https://plugins.octoprint.org/plugins/timelapseplus/](url)

monkeymademe, 2024-04-05:
@gatonero give this a go - https://github.com/monkeymademe/picamera2-WebUI-Lite/commit/9ab3fa8502a8887798568ac95fe6abc4707f16d4

monkeymademe, 2024-04-05:
@gatonero sorry if you already tried I made a mistake... https://github.com/monkeymademe/picamera2-WebUI-Lite/commit/7503dcdda0c4da99512e443d69b6c88861a00724 this will fix it. Basically I put the snapshot image in the gallery and the gallery did not like that (timestamp issues) so moved the snapshot to a subdirectory

gatonero, 2024-04-05:
I tested http://192.168.178.213:8080/snapshot in octoprint, with curl and with the browser. The result is `The requested URL was not found on the server. If you entered the URL manually please check your spelling and try again.`

monkeymademe, 2024-04-07:
I tried CURL on my own system and it worked. Just to check did you pull from main (not the release version)?

Could you tell me when you try to curl what the server log says?

Thanks for pointing out the URL I made the change

monkeymademe, 2024-04-09:
But if you open this page manually http://192.168.178.213:8080/snapshot  in browser its still not showing an image?

gatonero, 2024-04-09:
> But if you open this page manually http://192.168.178.213:8080/snapshot in browser its still not showing an image? 

It only says "Not Found"

I tested http://192.168.178.213:8080/video_feed in the snapshot settings of octoprint. It reported "Snapshot test failed. Could not get the snapshot URL, please check". It seems that there is an other algorithm implemented.

gatonero, 2024-04-09:
These are the default URL settings in octoprint:

- Stream URL: /webcam/?action=stream (no leading http, it's a relative path, it's correct this way!)
- Snapshot URL: http://localhost:8080/?action=snapshot

monkeymademe, 2024-04-09:
I am going to install octoprint and take a look

nvtkaszpir, 2024-05-02:
```shell
curl -I http://192.168.1.50:8080/snapshot
HTTP/1.1 200 OK
Server: Werkzeug/2.2.2 Python/3.11.2
Date: Thu, 02 May 2024 23:02:55 GMT
Content-Disposition: inline; filename=pimage_snapshot.jpg
Content-Type: image/jpg
Content-Length: 435374
Last-Modified: Thu, 02 May 2024 23:02:53 GMT
Cache-Control: no-cache
ETag: "1714690973.0288243-435374-1205607668"
Date: Thu, 02 May 2024 23:02:55 GMT
Connection: close

```
I believe the response should be `Content-Type: image/jpeg` (notice the missing 'e' letter)

PR #40 (2024-06-08): Fix /snapshot response to work with obico
Fixes #31
closed 2024-06-08 by monkeymademe