monkeymademe/CamUI #32: Strange error when trying to run on raspberry pi zero2W and 3A

lpanagis, 2024-03-22:
**Describe the bug**
Running "python app.py" gives me this error:
0:01:37.721844887] [1500]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:01:37.806086982] [1510]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media2 and ISP device /dev/media0
[0:01:37.806302353] [1510]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'

Camera Config:
{'controls': {'LensPosition': 1.0, 'ExposureValue': 0.0, 'ExposureTime': 220417486, 'AwbMode': 0, 'AfSpeed': 0, 'AfRange': 0, 'AeMeteringMode': 0, 'AeFlickerPeriod': 1000000, 'AeFlickerMode': 0, 'AeExposureMode': 0, 'AeEnable': True, 'AwbEnable': True, 'AeConstraintMode': 0, 'Brightness': 0.0, 'Contrast': 1.0, 'Saturation': 1.0, 'Sharpness': 1.0, 'AfMode': 0, 'ScalerCrop': [0, 0, 4608, 2592]}, 'rotation': {'hflip': 0, 'vflip': 0}, 'sensor-mode': 2, 'capture-settings': {'Resize': False, 'makeRaw': False, 'Resolution': 0, 'available-resolutions': [[4608, 2592], [2304, 1296], [1920, 1080], [1280, 720], [640, 360]]}}

Camera Settings:
{'Resize': False, 'makeRaw': False, 'Resolution': 0, 'available-resolutions': [[4608, 2592], [2304, 1296], [1920, 1080], [1280, 720], [640, 360]]}

Camera Set Resolution:
[4608, 2592]

[0:01:37.831955767] [1500]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-XBGR8888 (1) 640x480-SGBRG10_CSI2P
[0:01:37.832772614] [1510]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
[0:01:37.860837767] [1500]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-XBGR8888 (1) 1296x972-SGBRG10_CSI2P
[0:01:37.861606018] [1510]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
[0:01:37.887651475] [1500]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-XBGR8888 (1) 1920x1080-SGBRG10_CSI2P
[0:01:37.889522677] [1510]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA
[0:01:37.927825423] [1500]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-XBGR8888 (1) 2592x1944-SGBRG10_CSI2P
[0:01:37.928567320] [1510]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 2592x1944-SGBRG10_1X10 - Selected unicam format: 2592x1944-pGAA
Traceback (most recent call last):
  File "/home/lpanagis/picamera2-WebUI-Lite/app.py", line 44, in <module>
    video_config = picam2.create_video_configuration(main={'size':resolution}, sensor={'output_size': mode['size'], 'bit_depth': mode['bit_depth']})
TypeError: create_video_configuration() got an unexpected keyword argument 'sensor'

and the program exits
**To Reproduce**
run "python app.py"

**Expected behavior**
Picamera2 -WebUI-Lite should run

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: "Raspbian GNU/Linux 11 (bullseye)"

monkeymademe, 2024-03-24:
Its the v1 camera right?

I will check looks like something is amiss with the sensor parameter

monkeymademe, 2024-03-24:
I might see the issue, libcamera is out of date on bullseye and I heard there knock on issues with picamera2. I tried the v1 module on my setup (bookworm) on a pi4 and it was fine. I would recommend trying again with bookworm.

I am on the fence on if I would fix it so its more compatible cause its an issue that runs deep to the sensor modes.

lpanagis, 2024-03-26:
Thanks!
I installed the 32 bit version of Bookworm on my raspberry pi w2 and Pi3A+ and I get this error when trying to run it:
INFO:picamera2.picamera2:Camera configuration has been adjusted!
[0:20:56.078865427] [17222]  INFO Camera camera.cpp:1183 configuring streams: (0) 4608x2592-XBGR8888 (1) 1920x1080-SGBRG10_CSI2P
[0:20:56.079568258] [17236]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA
INFO:picamera2.picamera2:Configuration successful!
Traceback (most recent call last):
  File "/home/lpanagis/picamera2-WebUI-Lite/app.py", line 414, in <module>
    start_camera_stream()
  File "/home/lpanagis/picamera2-WebUI-Lite/app.py", line 282, in start_camera_stream
    picam2.configure(video_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1112, in configure
    self.configure_(camera_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1095, in configure_
    self.allocator.allocate(libcamera_config, camera_config.get("use_case"))
  File "/usr/lib/python3/dist-packages/picamera2/allocators/dmaallocator.py", line 43, in allocate
    fd = self.dmaHeap.alloc(f"picamera2-{i}", stream_config.frame_size)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/dma_heap.py", line 98, in alloc
    ret = fcntl.ioctl(self.__dmaHeapHandle.get(), DMA_HEAP_IOCTL_ALLOC, alloc)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 12] Cannot allocate memory
INFO:picamera2.picamera2:Camera closed successfully.

Is there not enough memory to run it? Should there be a correction declaring that these models are NOT compatible with picamera2-webui-lite?

the camera still works fine with libcamera.

monkeymademe, 2024-03-29:
Maybe related to this? https://github.com/monkeymademe/picamera2-WebUI-Lite/issues/25

closed 2024-07-10 by monkeymademe