monkeymademe/CamUI #33: Running in background or  @ PI start up and keep running

stovedoctor, 2024-04-08:
Current setup
RPI4 
Raspberry Pi OS with desktop
Release date: March 15th 2024
System: 64-bit
Kernel version: 6.6
Debian version: 12 (bookworm)
Camera module 3
wifi hotspot 
Would like to have run in background or not close when closing SSH terminal 
I want to use this as a rear view cam in a RV trailer and view feed from Device in the tow vehicle connected via PI's hotspot with no internet connection.
I have all that working but I really need to be able to keep it running or be able to run the app.py via a remote SSH terminal. As of now can do this but when closing the terminal or loosing connected the app stops.  As far I my limited skills are there is no way i have found to keep the app running after remote terminal is closed  and keep running other than physical connection via keyboard and monitor to the RPI4.

I have tried to have the pi run the app at start up but not sure why it will not start app.  by adding it to
/etc/rc.local
I may be doing it wrong 
`sudo python /home/rod/picamera2-WebUI-Lite-main/app.py &`

It might because running the  if you try to run the app with a command line outside of the directory there is a error
`python /home/rod/picamera2-WebUI-Lite-main/app.py`
command `python app.py` works fine if you are in the `picamera2-WebUI-Lite-main` directory 

```rod@casitapi2:/ $ python /home/rod/picamera2-WebUI-Lite-main/app.py
[0:20:38.989794380] [2068]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[0:20:39.021644526] [2071]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:20:39.023978579] [2071]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media2 and ISP device /dev/media1
[0:20:39.024122799] [2071]  INFO RPI pipeline_base.cpp:1144 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:20:39.026596127] [2068]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[0:20:39.057218386] [2074]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:20:39.059448978] [2074]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media2 and ISP device /dev/media1
[0:20:39.059572809] [2074]  INFO RPI pipeline_base.cpp:1144 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Traceback (most recent call last):
  File "/home/rod/picamera2-WebUI-Lite-main/app.py", line 22, in <module>
    with open("camera-config.json", "r") as file:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'camera-config.json'
```
Please excuse my ignorance if this is damp request or question for I am not a proficient user of the Pi interface and shell. 
just tiring my best with google searches to find a solution to using the PI as a rearview camera.

monkeymademe, 2024-04-09:
No worries there are no damp questions. 

1st address the RC Local might not work out because there can be issues with starting flask before the camera interface and network interface are ready. While for a quick run at boot its ok, I prefer to run either manually or as a service which I get to in a moment. 

2nd the error you have seems there is a missing file. Can you make sure you pull the latest from the repo... Make sure camera-config.json file is in the same directory as the app.py

At this point maybe the file was missing and pulling the latest work and RC.local was ok to start on boot then you should be ok... If not... 

Now we get to how to make service - Remove the RC local entry 1st

- Run the following command and note down the location for python `which python` should look like "/usr/bin/python"
- Goto the following directory `cd /etc/systemd/system/`
- Create and edit the following file `sudo nano picamera2-webui-lite.service`
- Paste this into the file, I made the changes for your username but if you see the line "ExecStart" the 1st part should be the result of your "which python" command we did at the start (if its the same then its all good)
`[Unit]
Description=Picamera2 WebUI Lite Server
After=network.target
[Service]
Type=simple
ExecStart=/usr/bin/python /home/pi/picamera2-WebUI-Lite/app.py
Restart=always
[Install]
WantedBy=multi-user.target`
- Save the file
- Run `sudo systemctl start picamera2-webui-lite.service`
- Run the following to check the service is running `sudo systemctl status picamera2-webui-lite.service`
- Run the following to enable the service to its running on reboot `sudo systemctl enable picamera2-webui-lite.service`

That 'should' be it. There might be a hicup with virtenv but I am not using that so I am not sure lets try those things to see if it works 1st. If the service does not run I will take another look.

stovedoctor, 2024-04-09:
zThanks for responding and willing to help Lol my typo Damp. Should have been dumb. 
I will give it a go when I get home. 
But a few questions. 

Btw I have tried a fresh pi image with nothing but your program running least 4 times on a fresh image.

I followed all the instructions  from the toms hardware page 
https://www.tomshardware.com/raspberry-pi/how-to-control-your-raspberry-pi-camera-using-a-web-ui.
Maybe I'm pulling from the wrong package?
From Toms write up.
`wget https://github.com/monkeymademe/picamera2-WebUI-Lite/archive/refs/heads/main.zip`

From the links on Toms page is how I installed picam2 UI.
But In the write up it shows 
Directory `picamera2-WebUI-Lite`

Your package gave the program directory a name of `picamera2-WebUI-Lite-main`

 Step 6 on toms writeup...
6. Change directory to Picamera 2.

`cd picamera2-WebUI-Lite`

When it errored no directory I listed the `user/rod ` and it showed your package created the directory as picamera2-WebUI-Lite-main`

I even tried renaming it removing the `-main` just in case things were referring back to the directory named that way.

The error for not being able to load the Json config only comes up if you try to load the app.py outside of the 'picamera2-WebUI-Lite` directory..

I ask about the directory name because of your new instructions of referring to `ExecStart=/usr/bin/python /home/pi/picamera2-WebUI-Lite/app.py`

Without -main in the directory name.

```
[Unit] Description=Picamera2 WebUI Lite Server After=network.target [Service] Type=simple ExecStart=/usr/bin/python /home/pi/picamera2-WebUI-Lite/app.py Restart=always [Install] WantedBy=multi-user.target```

Anyway I have a fresh install now 
Working but would like it to run in background.

Thanks again for responding.
I'm just now RE learning using Linux and Python commands  after over 15 years of not using. Back in the day I ran a remotely hosted colbolt appliances webserver for my websites and shopping online shopping carts and it's slowly coming back to me because I'm old.

stovedoctor, 2024-04-09:
Btw I'm trying to get this going because we are leaving for a 7 week road trip in a few weeks. 
Maybe there is a better way to make a rear view cam but I like the way your interface for controls works and looks. 
https://www.facebook.com/profile.php?id=61550792727679&mibextid=ZbWKwL

stovedoctor, 2024-04-09:
I just looked at the linked files on toms page it looks like a archive. 
So I will remove and install you're latest release. 
`
https://github.com/monkeymademe/picamera2-WebUI-Lite.git`

Will I need to install flask with the bookworm pi os ?

monkeymademe, 2024-04-09:
Could you check (just run the server before installing)... I seem to remember something recently changed in bookworm that improved support. either flask or something else.  Sorry not near a pi to check

monkeymademe, 2024-04-09:
@stovedoctor shoot me an email what your plans are maybe I can throw a timelapse feature (something thats simple start and stop) for you... sweeten the trip?

monkeymademe, 2024-04-09:
Also just pushed a change. I figured out the issue with the missing json file... Running as a service the script does not like the relative path (unless its directly a flask thing) so pulling in files next to the app.py it was just lost. So I put in some changes to detect the absolute path.

systemctl method of loading the server works now.

monkeymademe, 2024-04-09:
I will close this issue for now

closed 2024-04-09 by monkeymademe

stovedoctor, 2024-04-10:
Got it to work.on boot  with some fumbling around.....

 I copied your instructions.
errors then I noticed you had my old directory name with `-main`I fixed that l and still error
then I noticed you had the file path in twice
`ExecStart=/usr/bin/python /home/rod/picamera2-WebUI-Lite/picamera2-WebUI-Lite/app.py`
 . LOL Im glad im not the only one making stupid mistakes 
Thanks for the help 
You rock.. 
I also sent you an email

stovedoctor, 2024-04-30:
[image]

[image]