monkeymademe/CamUI #34: Add run as service option

codemonkey2k5, 2024-04-15:
Please add an option to allow the stream to run on Pi startup.   

Use case:  I am using this to monitor a 3D printer and have the PI power on when the Printer is powered on.  

Currently, I have to SSH into the PI and run the process.   If I close Putty, my video stream ends.   This Pi is headless and has no GUI installed.

monkeymademe, 2024-04-15:
Its fairly easy to set this up as a service:

- Run the following command and note down the location for python which python should look like "/usr/bin/python"
- Goto the following directory `cd /etc/systemd/system/`
- Create and edit the following file `sudo nano picamera2-webui-lite.service`
- Paste this into the file, in the line "ExecStart" the 1st part should be the result of your "which python" command we did at the start (if its the same then its all good) the 2nd path is the location of the cloned repo with the app.py

[Unit]
Description=Picamera2 WebUI Lite Server
After=network.target
[Service]
Type=simple
ExecStart=/usr/bin/python /home/pi/picamera2-WebUI-Lite/app.py
Restart=always
[Install]
WantedBy=multi-user.target

- Save the file
- Run `sudo systemctl start picamera2-webui-lite.service`
- Run the following to check the service is running` sudo systemctl status picamera2-webui-lite.service`
- Run the following to enable the service to its running on reboot `sudo systemctl enable picamera2-webui-lite.service`

An alt to this if you don't want to run the service (maybe you want to see the log for debugging or the likes) you can install something like screen `sudo apt install screen` and run it like this `screen python app.py` then it will run it in the background. Then you want to get back to it you can log back in to the system and then run `screen -x` and it should kick back into the server

codemonkey2k5, 2024-04-17:
Just wanted to thank you again.  Got it running as a service.

monkeymademe, 2024-04-17:
You are very welcome. I have put the instructions for making a service on the readme and I will close this issue now

closed 2024-04-17 by monkeymademe