monkeymademe/CamUI #37: Add frame rate option to time lapse

codemonkey2k5, 2024-04-19:
It would be very helpful if you could designate how many images per sec/min/hour the screen capture option takes. 
This would help for those with limited storage space.   For days long prints, 1 frame per 5 min may be enough.

monkeymademe, 2024-04-19:
If you are talking about the timelapse feature... thats totally far from unfinished... what I put in was a quick hack for an event. This is lowish in the prio to fix.

BUT... 

If you fancy a little dig in the code in line 351 of the app.py file there is a variable `interval = 2` which is in seconds so edit that and the interval will change

codemonkey2k5, 2024-04-19:
Yep and thanks for the insight.   Appreciate the quick response.

monkeymademe, 2024-07-03:
I am closing this ticket because timelapse is an unfinished feature and will not be in the next major release. But will be implemented after that (1.0.0 will be pi5 support and 1.0.1,2,3 etc will be fixes 1.1.0 will be timelapse... or there abouts)

Don't worry timelapse is important and will be there

closed 2024-07-03 by monkeymademe