monkeymademe/CamUI #38: Remove Lite from the name

monkeymademe, 2024-04-21:
this thing is turning into a monster.... maybe just forget the idea of lite and just have one project

geerlingguy, 2024-06-08:
Heh for sure. I wasn't expecting much at all based on the name.

Then I downloaded it, ran it... and realized this is the nicest web camera GUI I've ever used on a Pi! Setup was a breeze too (even though I was on Lite Pi OS).

geerlingguy, 2024-06-19:
@monkeymademe - FYI, I hope to use this project in a video I'm working on next week — if you change the name of the project by Monday that'd be great in terms of me mentioning it as just "picamera2-WebUI" :)

monkeymademe, 2024-06-19:
Aw man thats awesome! Actually I am working on the next version right now which will be a huge change in how the UI will work... I will get my skates on to get it up... In terms of the name I am dropping the lite for sure...

monkeymademe, 2024-06-19:
@geerlingguy - Aw man thats awesome! Actually I am working on the next version right now which will be a huge change in how the UI will work... I will get my skates on to get it up... In terms of the name I am dropping the lite for sure...

geerlingguy, 2024-06-19:
That's awesome! And I can't remember if it was @Tschucker or maybe the booth next to his running an 8mm film scanner—one of them was using this same utility for a web UI for the camera in one demo at Open Sauce!

monkeymademe, 2024-07-03:
'Lite' has been removed

closed 2024-07-03 by monkeymademe