monkeymademe/CamUI #39: Expert Mode or somethin

TimPGamer, 2024-04-22:
My Idea is to implement some kind of expert mode or an addon, where u have a variety of options, where u can configure like a servo, for moving the camera manually and configure the Pins on which the servo will be controlled. Another feature would be to like add a manual focus, where u can control an external motor, connected to the lens, to control the focus with this motor, as not all camerad have a macanical focus. this would then be again controlled by a selected GPIO.

I know this is really much to ask for, but it would be nice to have. If you dont want to add this, you could at least make it possible to add addons (like minecraft mods, just add and go) and let us know how.

monkeymademe, 2024-07-19:
Sorry I hear you... something like this is planned

monkeymademe, 2025-04-10:
v2.0.0 has removed GPIO features as I want to rework them into something better... Pretty much your expert mode. It will happen soon I hope

TimPGamer, 2025-04-17:
No problem. I can wait. Take your time to implement it correctly