monkeymademe/CamUI #42: OSError: [Errno 12] Cannot allocate memory

jphellemons, 2024-05-29:
**Describe the bug**
Just downloaded and extracted the zip on my pi3 b+ and got this message. streaming works over LAN. Hope I can get this software to work as it seems really awesome!

**To Reproduce**
downloaded the zip. extracted and ran python app.py
this is on a rpi 3
`cat /etc/debian_version` gives 12.5

```
libcamera-vid --list-cameras
Available cameras
-----------------
0 : ov5647 [2592x1944 10-bit GBRG] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
                             1296x972 [43.25 fps - (0, 0)/2592x1944 crop]
                             1920x1080 [30.62 fps - (348, 434)/1928x1080 crop]
                             2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]
```

**Expected behavior**
I expected it to run and not crash so that I could open the webapp on a browser

**Screenshots**
console output is probably better:
```
[0:31:59.202068590] [1536]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[0:31:59.248394019] [1539]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:31:59.252915457] [1539]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[0:31:59.253028375] [1539]  INFO RPI pipeline_base.cpp:1102 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:31:59.257353665] [1536]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[0:31:59.302053771] [1542]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:31:59.306376666] [1542]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[0:31:59.306575157] [1542]  INFO RPI pipeline_base.cpp:1102 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'

Camera Config:
{'controls': {'LensPosition': 1.0, 'ExposureValue': 0.0, 'ExposureTime': 220417486, 'AwbMode': 0, 'AfSpeed': 0, 'AfRange': 0, 'AeMeteringMode': 0, 'AeFlickerPeriod': 1000000, 'AeFlickerMode': 0, 'AeExposureMode': 0, 'AeEnable': True, 'AwbEnable': True, 'AeConstraintMode': 0, 'Brightness': 0.0, 'Contrast': 1.0, 'Saturation': 1.0, 'Sharpness': 1.0, 'AfMode': 0, 'ScalerCrop': [0, 0, 4608, 2592]}, 'rotation': {'hflip': 0, 'vflip': 0}, 'sensor-mode': 0, 'capture-settings': {'Resize': False, 'makeRaw': False, 'Resolution': 0, 'available-resolutions': [[4608, 2592], [2304, 1296], [1920, 1080], [1280, 720], [640, 360]]}}

Camera Settings:
{'Resize': False, 'makeRaw': False, 'Resolution': 0, 'available-resolutions': [[4608, 2592], [2304, 1296], [1920, 1080], [1280, 720], [640, 360]]}

Camera Set Resolution:
[4608, 2592]

[0:31:59.322196517] [1536]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 640x480-SGBRG10_CSI2P
[0:31:59.322853344] [1542]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
[0:31:59.342930100] [1536]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 1296x972-SGBRG10_CSI2P
[0:31:59.343579011] [1542]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
[0:31:59.362566644] [1536]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 1920x1080-SGBRG10_CSI2P
[0:31:59.363198627] [1542]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA
[0:31:59.385495815] [1536]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 2592x1944-SGBRG10_CSI2P
[0:31:59.386139570] [1542]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 2592x1944-SGBRG10_1X10 - Selected unicam format: 2592x1944-pGAA

Video Config:
{'use_case': 'video', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'Rec709'>, 'buffer_count': 6, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (4608, 2592)}, 'lores': None, 'raw': {'format': 'SGBRG10_CSI2P', 'size': (4608, 2592)}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (33333, 33333)}, 'sensor': {'output_size': (640, 480), 'bit_depth': 10}, 'display': 'main', 'encode': 'main'}

Picamera2 Camera Properties:
{'Model': 'ov5647', 'UnitCellSize': (1400, 1400), 'ColorFilterArrangement': 2, 'Location': 2, 'Rotation': 0, 'PixelArraySize': (2592, 1944), 'PixelArrayActiveAreas': [(16, 6, 2592, 1944)], 'ScalerCropMaximum': (0, 0, 2592, 1944), 'SystemDevices': (20749, 20737, 20738, 20739), 'SensorSensitivity': 1.0}

INFO:picamera2.picamera2:Camera configuration has been adjusted!
[0:31:59.462097291] [1536]  INFO Camera camera.cpp:1183 configuring streams: (0) 4608x2592-XBGR8888 (1) 640x480-SGBRG10_CSI2P
[0:31:59.462659326] [1542]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
INFO:picamera2.picamera2:Configuration successful!
Traceback (most recent call last):
  File "/home/tweet/picamera2-WebUI-Lite-main/app.py", line 508, in <module>
    start_camera_stream()
  File "/home/tweet/picamera2-WebUI-Lite-main/app.py", line 304, in start_camera_stream
    picam2.configure(video_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1114, in configure
    self.configure_(camera_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1097, in configure_
    self.allocator.allocate(libcamera_config, camera_config.get("use_case"))
  File "/usr/lib/python3/dist-packages/picamera2/allocators/dmaallocator.py", line 43, in allocate
    fd = self.dmaHeap.alloc(f"picamera2-{i}", stream_config.frame_size)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/dma_heap.py", line 98, in alloc
    ret = fcntl.ioctl(self.__dmaHeapHandle.get(), DMA_HEAP_IOCTL_ALLOC, alloc)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 12] Cannot allocate memory
INFO:picamera2.picamera2:Camera closed successfully.
```

**Desktop (please complete the following information):**
 Raspbian latest afaik. logged in by ssh from microsoft terminal

**Additional context**
might be related to https://github.com/monkeymademe/picamera2-WebUI-Lite/issues/32

I can stream the video to VLC (client) with this command:
`rpicam-vid -t 0 --inline -o - | cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554/stream1}' :demux=h264`
just to verify that the cam works and the networking etc. Might just be the case that my pi is not powerfull enough? **rpi 3 b+**

jphellemons, 2024-05-31:
Seems related to https://github.com/raspberrypi/picamera2/issues/972#issuecomment-1980573868

gajdipajti, 2024-07-05:
Hi, 
I ran into the same problem with my Raspberry Pi 2B + v1 camera. I could only start the software after changing the **Resolution** from 0 to 2 in **camera-config.json**. With 0 I ran into the same memory allocation error. 
Maybe 1920x1080 (2) can be the default setting.

```
cat /etc/debian_version
12.1
```

```
$ libcamera-vid --list-cameras
Available cameras
-----------------
0 : ov5647 [2592x1944 10-bit GBRG] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [30.00 fps - (65535, 65535)/65535x65535 crop]
                             1296x972 [30.00 fps - (65535, 65535)/65535x65535 crop]
                             1920x1080 [30.00 fps - (65535, 65535)/65535x65535 crop]
                             2592x1944 [30.00 fps - (65535, 65535)/65535x65535 crop]
```

jmw0149, 2024-07-07:
Ran into the same problem. Solved using buffer_count=2 in create_video_configuration suggested by: https://github.com/raspberrypi/picamera2/issues/972#issuecomment-1980573868
No buffer_count is included in the source code.
Raspberry Pi 3B+ ver 1.4

monkeymademe, 2024-07-08:
Yeah this is maybe something I could setup in a setting but its very breakable... picamera2's Video config requests a buffer count of 6 by default and anything lower than that can cause stream issues (jitter, missed frames) I experimented with that before and got really bad performance. 

I would recommend using a pi4 or better. But in the interest of trying to keep all pi3 happy and the pi Zero2 I would maybe make this a configurable setting with huge warnings and an are you sure popup.... its in the roadmap for the next version

monkeymademe, 2024-07-08:
Going to close this issue cause the buffer_count will be addressed in this feature request: https://github.com/monkeymademe/picamera2-WebUI/issues/51

closed 2024-07-08 by monkeymademe