monkeymademe/CamUI #43: Slight differences for install on Lite version of Pi OS

geerlingguy, 2024-06-08:
**Describe the bug**

When I follow the official README, it assumes I'm running full desktop Pi OS, and some things aren't present on the 'Lite' version which require installation before the WebUI will run.

**To Reproduce**

  1. Flash a fresh image of Pi OS 12 'Lite' to a microSD card.
  2. Follow the installation instructions in the README.

**Expected behavior**

Everything works.

**Additional context**

See: https://github.com/geerlingguy/3d-printing/issues/9#issuecomment-2155786176

A few things I had to do to get things working:

  1. apt install `git` and `python3-pip`
  2. apt install `python3-picamera2 --no-install-recommends`
  3. pip install `Flask` and `Pillow`

Note that I dealt with the whole 'externally managed' thing by nuking the `EXTERNALLY-MANAGED` file so I could just install things on system python. I tried using a `venv` in the project inside which I tried installing all the dependencies, but it complained about `picamera2` not being available, even though I had the system level package `python3-picamera2` installed... So I just deleted the virtual Python environment and stomped all over my system install :)

davidhorsell, 2024-08-14:
I tried this on a Pi Zero 2 with Pi OS Lite 12 and a Pi Global Shutter camera attached. Communication to a Windows PC through the USB (RNDIS ethernet gadget) rather than WiFi. The sequence that worked for me was:

$ sudo apt install git
$ sudo apt install python3-flask
$ sudo apt install python3-pillow
$ sudo apt install python3-picamera2

(I also could not get it to work if installing via pip.) After that, the git clone and running app.py works fine (so far).

Mraanderson, 2025-11-07:
This did not want to run on a Pi Zero W so I settled for a 3B+. No venv and will add the auto start by system d once I confirm working with the v1.3 camera. I just want to power on with this lump attached to my telescope and start snapping lunar and planetary images. Stitching with Microsoft ICE after download to a PC. Headless Trixie. 

pi@Pi-Cam:~/picamera2-WebUI $ history
    1  sudo apt update &&  sudo apt full-upgrade -y
    2  sudo apt autoremove &&  sudo apt autoclean
    3  sudo apt install git pip python3-flask
    4  sudo apt install python3-picamera2 --no-install-recommends
    5  git clone https://github.com/monkeymademe/picamera2-WebUI.git
    6  cd picamera2-WebUI
    7  python app.py