monkeymademe/CamUI #44: Allow setting defaults?

geerlingguy, 2024-06-10:
**Is your feature request related to a problem? Please describe.**

I use this UI for checking up on 3D prints when I'm away, but I shut down the Pi when I'm not using the 3D Printer. I would like to make it so when I boot up the Pi and launch the app, it uses my default settings for things like manual focus, auto exposure, and resolution.

Right now using a Camera Module v3 it always defaults to 4K resolution, which is choppy, and autofocus, which gets a bit confused depending on how my print is going.

**Describe the solution you'd like**

I'd love to be able to set a few things in a YAML or JSON config file like `defaults.json` (which is not tracked in this repo, so I could write my own file within... or maybe a file path I could pass in when running `python app.py`?).

**Describe alternatives you've considered**

Manually clicking through settings every time I restart my Pi.

**Additional context**

N/A

geerlingguy, 2024-06-10:
Ah... just found this on the main screen:

[image]

When I click save, it overwrites `camera-config.json` with the new values.

It may be good to have a set of repository defaults ('camera-defaults.json', maybe?), and the application could merge in any defaults from a `camera-config.json` file if it exists—and add `camera-config.json` to `.gitignore` so a user could save their own config without overriding the git tracked file.

monkeymademe, 2024-07-03:
This is considered for the next release

monkeymademe, 2024-07-08:
This has been improved in the Beta pre release and could use some testing - https://github.com/monkeymademe/picamera2-WebUI/releases/tag/v0.1.0-beta

monkeymademe, 2024-07-19:
This was released in v1.0.0 but also had issues or suggestions... latest release should be what your looking for @geerlingguy

closed 2024-07-19 by monkeymademe