monkeymademe/CamUI #47: set navigation active state based on current page

dmnkhhn, 2024-07-03:
Hello!

I have made some changes to my local install to make the active state of the navigation dynamic.
Right now the the "active" class and the `aria-current="page"` attribute is always set on the first page only.

You have stated in the release notes that there are major changes coming to how this app works.
My change requires changes in `app.py` and `base.html` - are you interested in pull request or will this become obsolete?

monkeymademe, 2024-07-03:
If you would like you can check out the beta branch thats going to be the next release... there is a lot of changes in the menu to account for more than one camera (pi5)

monkeymademe, 2024-07-03:
I have to admit I was not sure what you meant at first and then I took a look and realized the nav menu... so I implemented your suggestion into the beta.

For the main you are welcome to pull request away and I will merge it... I'll close this ticket if you feel thats enough

dmnkhhn, 2024-07-04:
Will do so, thanks!

closed 2024-07-04 by dmnkhhn