monkeymademe/CamUI #52: Manually running app.py works, service does not

ReaderGuy42, 2024-07-09:
**Describe the bug**
If I run app.py I get the web UI, but when I try running it as a service it doesn't 

**To Reproduce**
Steps to reproduce the behavior:
1. Create " /etc/systemd/system/picamera2-webui-lite.service" file
2. Enter details, from Github description:
```
[Unit]
Description=Picamera2 WebUI Lite Server
After=network.target
[Service]
Type=simple
ExecStart=/usr/bin/python /home/mediapi/picamera2-WebUI/app.py
Restart=always
[Install]
WantedBy=multi-user.target
```
3. Run ` sudo systemctl start picamera2-webui-lite.service`, `systemct daemon-reload`, `sudo systemctl enable picamera2-webui-lite.service`
4. Reboot
5. `sudo systemctl status picamera2-webui-lite.service`, results in error:

```
picamera2-webui-lite.service - Picamera2 WebUI Lite Server
     Loaded: loaded (/etc/systemd/system/picamera2-webui-lite.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Tue 2024-07-09 16:19:49 CEST; 5min ago
   Duration: 57ms
    Process: 935 ExecStart=/usr/bin/python /home/mediapi/picamera2-WebUI/app.py (code=exited, status=2)
   Main PID: 935 (code=exited, status=2)
        CPU: 57ms

Jul 09 16:19:48 rpcam systemd[1]: picamera2-webui-lite.service: Scheduled restart job, restart counter is at 4.
Jul 09 16:19:48 rpcam systemd[1]: Stopped picamera2-webui-lite.service - Picamera2 WebUI Lite Server.
Jul 09 16:19:48 rpcam systemd[1]: Started picamera2-webui-lite.service - Picamera2 WebUI Lite Server.
Jul 09 16:19:48 rpcam systemd[1]: picamera2-webui-lite.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Jul 09 16:19:48 rpcam systemd[1]: picamera2-webui-lite.service: Failed with result 'exit-code'.
Jul 09 16:19:49 rpcam systemd[1]: picamera2-webui-lite.service: Scheduled restart job, restart counter is at 5.
Jul 09 16:19:49 rpcam systemd[1]: Stopped picamera2-webui-lite.service - Picamera2 WebUI Lite Server.
Jul 09 16:19:49 rpcam systemd[1]: picamera2-webui-lite.service: Start request repeated too quickly.
Jul 09 16:19:49 rpcam systemd[1]: picamera2-webui-lite.service: Failed with result 'exit-code'.
Jul 09 16:19:49 rpcam systemd[1]: Failed to start picamera2-webui-lite.service - Picamera2 WebUI Lite Server.
~
```

**Expected behavior**
Launch web UI on boot

**Desktop (please complete the following information):**
 - OS: Raspberry Pi OS (from July 4th)
 - Browser: Firefox
 - Version: 0.0.5

monkeymademe, 2024-07-09:
Sorry I totally forgot when removing the lite from the app the service directory names and service names would change also... I have made the change now

https://github.com/monkeymademe/picamera2-WebUI/commit/cab50a8cdbe70f2f85b4acd1678da754628da012

Basically where ever is says lite... remove lite. (the former plan was to have a feed only app with basic functions and have a pro big version for all the fancy stuff... plans changed and there will be one app now)

monkeymademe, 2024-07-09:
Let me know if that works and sorry for the inconvenience

ReaderGuy42, 2024-07-09:
Ah cool, that looks good, but I'm somehow getting the same error message:

[image]

monkeymademe, 2024-07-09:
I am afk at the moment so I can't run the code my myself but it's possible if you change the service then run the command again the old service might be running... I am not as confident with services to be honest. I would maybe try the ye olde reboot and see if that kicks it into gear otherwise I'll try when I am available to see what the issue might be... can you tell me also if you are in the main or beta branch

monkeymademe, 2024-07-10:
Did the reboot work?

ReaderGuy42, 2024-07-10:
No, unfortunately, several reboots didn't change anything

monkeymademe, 2024-07-10:
So either there is something wrong with the app.py thats not able to run - can be that the camera is busy or not detected or there is a permission issue with that directory you used to store the app... you cloned with 'git clone ***URL'? or with sudo?

monkeymademe, 2024-07-10:
Maybe paste here, whats in the service file too

monkeymademe, 2024-07-10:
Ah also what OS are you using and you said you used the app.py manually and it worked fine?

monkeymademe, 2024-07-10:
ok last try - can you run `journalctl -u picamera2-webui.service` then paste me the output... also double check there is not more than one picamera2-WebUI service running

monkeymademe, 2024-07-12:
Any luck?

monkeymademe, 2024-07-19:
Please reopen if this is still an issue

closed 2024-07-19 by monkeymademe