monkeymademe/CamUI #59: Web Server Port Number

dalogue1, 2024-07-12:
I am running the picamera2-WebUI on the same machine as another application with a Web Server (i.e. BirdNET-Pi) and it already is using port 8080.  It would be nice to pass the port number into the app.py as an argument and to default to 8080 if the argument was not provided. 

Not a big deal because it is literally the final line of the script and is easy to change.

monkeymademe, 2024-07-13:
This is an easy fix I will look at it on the next release (very soon)

monkeymademe, 2024-07-16:
Give this ago - if ok then I will close this ticket

https://github.com/monkeymademe/picamera2-WebUI/commit/507a0e5ff80fa1433760e4ec8556491f40fa967b

monkeymademe, 2024-07-16:
Released to https://github.com/monkeymademe/picamera2-WebUI/releases/tag/v1.0.1

closed 2024-07-16 by monkeymademe

dalogue1, 2024-07-17:
It works great!