monkeymademe/CamUI #66: Crash / fail to start with RPi official global scan cam module

snowrodeo, 2024-07-17:
Very vanilla install of bookworm and RPi5.  System all updated, no joy.

```
erik@rpi5-camera1:~/picamera2-WebUI $ python app.py 
[0:10:51.186758361] [9697]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+65-6ddd79b5
[0:10:51.194760788] [9704]  INFO RPI pisp.cpp:695 libpisp version v1.0.6 b567f0455680 17-06-2024 (10:20:00)
[0:10:51.204216116] [9704]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx296@1a to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
/home/erik/picamera2-WebUI/static/camera_config

Detected Cameras:
[{'Model': 'imx296', 'Location': 2, 'Rotation': 180, 'Id': '/base/axi/pcie@120000/rp1/i2c@88000/imx296@1a', 'Num': 0}]

Camera Info:
{'Model': 'imx296', 'Location': 2, 'Rotation': 180, 'Id': '/base/axi/pcie@120000/rp1/i2c@88000/imx296@1a', 'Num': 0}

{'Num': 0, 'Model': 'imx219', 'Is_Pi_Cam': True, 'Has_Config': False, 'Config_Location': 'default_imx219.json'}
0 0 and imx296  imx219
{'Num': 1, 'Model': 'imx708', 'Is_Pi_Cam': True, 'Has_Config': True, 'Config_Location': 'Testing.json'}
0 1 and imx296  imx708
Camera config has changed since last boot - Adding new Camera
[0:10:51.206082496] [9697]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+65-6ddd79b5
[0:10:51.214400532] [9707]  INFO RPI pisp.cpp:695 libpisp version v1.0.6 b567f0455680 17-06-2024 (10:20:00)
[0:10:51.224589783] [9707]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx296@1a to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[0:10:51.225780722] [9697]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+65-6ddd79b5
[0:10:51.234023277] [9710]  INFO RPI pisp.cpp:695 libpisp version v1.0.6 b567f0455680 17-06-2024 (10:20:00)
[0:10:51.243742307] [9710]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx296@1a to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
picamera2.picamera2 INFO: Initialization successful.
picamera2.picamera2 INFO: Camera now open.
picamera2.picamera2 DEBUG: <libcamera._libcamera.CameraManager object at 0x7fff12d1e2b0>
[0:10:51.247041589] [9697]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
picamera2.picamera2 DEBUG: Requesting configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': {'format': 'BGGR_PISP_COMP1', 'size': (1456, 1088), 'stride': 1472, 'framesize': 1601536}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'sensor': {'bit_depth': 10, 'output_size': (1456, 1088)}, 'display': 'main', 'encode': 'main'}
picamera2.picamera2 INFO: Camera configuration has been adjusted!
[0:10:51.247742882] [9697]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 1456x1088-BGGR_PISP_COMP1
[0:10:51.247877974] [9710]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx296@1a - Selected sensor format: 1456x1088-SBGGR10_1X10 - Selected CFE format: 1456x1088-PC1B
picamera2.picamera2 INFO: Configuration successful!
picamera2.picamera2 DEBUG: Final configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': {'format': 'BGGR_PISP_COMP1', 'size': (1456, 1088), 'stride': 1472, 'framesize': 1601536}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'sensor': {'bit_depth': 10, 'output_size': (1456, 1088)}, 'display': 'main', 'encode': 'main'}
picamera2.picamera2 DEBUG: Streams: {'main': <libcamera._libcamera.Stream object at 0x7ffef93934b0>, 'lores': None, 'raw': <libcamera._libcamera.Stream object at 0x7ffef93934f0>}
picamera2 DEBUG: Allocated 4 buffers for stream 0 with fds [32, 35, 38, 41]
picamera2 DEBUG: Allocated 4 buffers for stream 1 with fds [44, 47, 50, 53]

Camera Settings:
{'Resize': False, 'makeRaw': False, 'Resolution': 0}

Camera Set Resolution:
(1456, 1088)

Traceback (most recent call last):
  File "/home/erik/picamera2-WebUI/app.py", line 448, in <module>
    camera_obj = CameraObject(camera_num, camera_info)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/erik/picamera2-WebUI/app.py", line 92, in __init__
    self.init_camera()
  File "/home/erik/picamera2-WebUI/app.py", line 200, in init_camera
    mode = self.camera.sensor_modes[self.sensor_mode]
           ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
IndexError: list index out of range
picamera2.picamera2 DEBUG: Camera was not started
picamera2.picamera2 INFO: Camera closed successfully.
picamera2.picamera2 DEBUG: Resources now free: <picamera2.picamera2.Picamera2 object at 0x7fff0aeb11d0>
```

monkeymademe, 2024-07-17:
My bad if you pull it will work now.

https://github.com/monkeymademe/picamera2-WebUI/commit/1ddb58ccba94e2552ba3193851ecb290a6d7617f

The issue was when a new camera is added there is some default values set including the sensor mode. For debugging an issue a while back I set the default to 1 meaning it would pick the 2nd sensor mode for a new camera. But the GS camera only has one sensor mode so it threw the error.

Thanks for letting me know

closed 2024-07-17 by monkeymademe

snowrodeo, 2024-07-17:
Confirmed working on my side!  Thanks for the quick fix!!