monkeymademe/CamUI #68: "FileNotFoundError: [Errno 2] No such file or directory" After Changing/Saving Settings and then a Reboot

kenj1986, 2024-07-17:
My apologies if I'm doing this wrong or if this is a duplicate, it's my first bug report.
After changing camera settings in the UI and saving them, app.py fails to run.
I suspect that app.py can't find the config json file that's created when the camera settings are saved.

Steps to reproduce the behavior:
1. Install picamera2-WebUI and all prerequisites
2. Execute app.py - works fine
3. Use UI to change and save camera settings
4. Restart app.py (or the whole machine)
5. Execute app.py - fail (see below dump)
Steps I've used to "fix" it:
7. Edit camera-last-config.json and change Has_Config to false
8. Execute app.py  - works, but settings are reverted to defaults (as you'd expect when telling it to ignore the config file)

Here's a dump of the output:
> [user]@[hostname]:~ $ python picamera2-WebUI/app.py
> [0:01:36.561546807] [863]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+65-6ddd79b5
> [0:01:36.676957503] [866]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
> [0:01:36.679252925] [866]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media4 and ISP device /dev/media0
> [0:01:36.679369579] [866]  INFO RPI pipeline_base.cpp:1104 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
> /home/[user]/picamera2-WebUI/static/camera_config
> 
> Detected Cameras:
> [{'Model': 'imx708', 'Location': 2, 'Rotation': 180, 'Id': '/base/soc/i2c0mux/i2c@1/imx708@1a', 'Num': 0}]
> 
> 
> Camera Info:
> {'Model': 'imx708', 'Location': 2, 'Rotation': 180, 'Id': '/base/soc/i2c0mux/i2c@1/imx708@1a', 'Num': 0}
> 
> {'Num': 0, 'Model': 'imx708', 'Is_Pi_Cam': True, 'Has_Config': True, 'Config_Location': '2024-07-17.json'}
> 0 0 and imx708  imx708
> Detected camera 0: imx708 matched last used in config.
> [0:01:36.683732818] [863]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+65-6ddd79b5
> [0:01:36.812353406] [869]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
> [0:01:36.814498240] [869]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media4 and ISP device /dev/media0
> [0:01:36.814576553] [869]  INFO RPI pipeline_base.cpp:1104 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
> [0:01:36.817084854] [863]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+65-6ddd79b5
> [0:01:36.944356224] [872]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
> [0:01:36.946493445] [872]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media4 and ISP device /dev/media0
> [0:01:36.946573981] [872]  INFO RPI pipeline_base.cpp:1104 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
> picamera2.picamera2 INFO: Initialization successful.
> picamera2.picamera2 INFO: Camera now open.
> picamera2.picamera2 DEBUG: <libcamera._libcamera.CameraManager object at 0x7f9257dc30>
> picamera2.picamera2 DEBUG: Requesting configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': {'format': 'SBGGR10_CSI2P', 'size': (1536, 864), 'stride': 1920, 'framesize': 1658880}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'sensor': {'bit_depth': 10, 'output_size': (1536, 864)}, 'display': 'main', 'encode': 'main'}
> picamera2.picamera2 INFO: Camera configuration has been adjusted!
> [0:01:36.955617602] [863]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 1536x864-SBGGR10_CSI2P
> [0:01:36.956106800] [872]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
> picamera2.picamera2 INFO: Configuration successful!
> picamera2.picamera2 DEBUG: Final configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': {'format': 'SBGGR10_CSI2P', 'size': (1536, 864), 'stride': 1920, 'framesize': 1658880}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'sensor': {'bit_depth': 10, 'output_size': (1536, 864)}, 'display': 'main', 'encode': 'main'}
> picamera2.picamera2 DEBUG: Streams: {'main': <libcamera._libcamera.Stream object at 0x7f925a1370>, 'lores': None, 'raw': <libcamera._libcamera.Stream object at 0x7f925a13b0>}
> picamera2 DEBUG: Allocated 4 buffers for stream 0 with fds [26, 29, 32, 35]
> picamera2 DEBUG: Allocated 4 buffers for stream 1 with fds [38, 41, 44, 47]
> picamera2.picamera2 DEBUG: Requesting configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': {'format': 'SBGGR10_CSI2P', 'size': (2304, 1296), 'stride': 2880, 'framesize': 3732480}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'sensor': {'bit_depth': 10, 'output_size': (2304, 1296)}, 'display': 'main', 'encode': 'main'}
> picamera2.picamera2 INFO: Camera configuration has been adjusted!
> [0:01:36.969486487] [863]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 2304x1296-SBGGR10_CSI2P
> [0:01:36.969910857] [872]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
> picamera2.picamera2 INFO: Configuration successful!
> picamera2.picamera2 DEBUG: Final configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': {'format': 'SBGGR10_CSI2P', 'size': (2304, 1296), 'stride': 2880, 'framesize': 3732480}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'sensor': {'bit_depth': 10, 'output_size': (2304, 1296)}, 'display': 'main', 'encode': 'main'}
> picamera2.picamera2 DEBUG: Streams: {'main': <libcamera._libcamera.Stream object at 0x7f925a1370>, 'lores': None, 'raw': <libcamera._libcamera.Stream object at 0x7f925a13b0>}
> picamera2 DEBUG: Allocated 4 buffers for stream 0 with fds [26, 29, 32, 35]
> picamera2 DEBUG: Allocated 4 buffers for stream 1 with fds [38, 41, 44, 47]
> picamera2.picamera2 DEBUG: Requesting configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': {'format': 'SBGGR10_CSI2P', 'size': (4608, 2592), 'stride': 5760, 'framesize': 14929920}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'sensor': {'bit_depth': 10, 'output_size': (4608, 2592)}, 'display': 'main', 'encode': 'main'}
> picamera2.picamera2 INFO: Camera configuration has been adjusted!
> [0:01:36.985746927] [863]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 4608x2592-SBGGR10_CSI2P
> [0:01:36.986177242] [872]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected unicam format: 4608x2592-pBAA
> picamera2.picamera2 INFO: Configuration successful!
> picamera2.picamera2 DEBUG: Final configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': {'format': 'SBGGR10_CSI2P', 'size': (4608, 2592), 'stride': 5760, 'framesize': 14929920}, 'controls': {'NoiseReductionMo [... cut ...]
>   File "/usr/lib/python3/dist-packages/picamera2/encoders/encoder.py", line 231, in start
>     self._start()
>   File "/usr/lib/python3/dist-packages/picamera2/encoders/v4l2_encoder.py", line 149, in _start
>     fcntl.ioctl(self.vd, VIDIOC_STREAMON, typev)
> FileNotFoundError: [Errno 2] No such file or directory
> picamera2.picamera2 INFO: Camera stopped
> picamera2.picamera2 INFO: Camera closed successfully.
> [user]@[hostname]:~ $

I'm using a Raspberry Pi 4 with 4 GB of RAM and I'm running Raspberry Pi OS Lite 64 bit.

monkeymademe, 2024-07-17:
Acutally can you tell me what files are in the static/camera_config folder and also paste in whats inside the file camera-last-config.json

monkeymademe, 2024-07-19:
Ok I tried to make the saving loading and using default config a little more robust so if a file is missing the default settings will be taken and the missing file is off the record

I hope this will solve the issue. If not please reopen the ticket

closed 2024-07-19 by monkeymademe