monkeymademe/CamUI #76: RuntimeError: can't start new thread

AleksanderKruecken, 2024-08-02:
Hi, 

the picamera2 webui stops working after a day or two.
Steps to reproduce: 
http://172.17.1.19:8080/ returns ERR_EMPTY_RESPONSE after working correctly for a day or two.

HW:
Raspberry Pi zero 2W

OS: Raspberry pi OS Bookworm (Debian GNU/Linux 12) with desktop GUI

Camera: 
Raspberry pi camera module 1

Browser:
Chrome

Version of app:
1.0.2-2-g34fc6be

Additional context
I'm watching the camera feed in my brower on 8080 port and as a picture glance card in home assistant OS.
The camera in Home assistant OS is added with the help of MJPEG IP Camera integration.

I'm attaching the log file for picamera2-webui service.
The first line shows the last working condition (HTTP status 200), after that the app crashed.

Aug 02 10:22:09 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:09] "GET /control_camera_0 HTTP/1.1" 200 -
Aug 02 10:22:09 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:09] "GET /control_camera_0 HTTP/1.1" 200 -
Aug 02 10:22:09 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:09] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
Aug 02 10:22:09 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:09] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
Aug 02 10:22:09 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:09] "GET /static/css/bootstrap-icons.css HTTP/1.1" 304 -
Aug 02 10:22:09 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:09] "GET /static/css/bootstrap-icons.css HTTP/1.1" 304 -
Aug 02 10:22:09 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:09] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
Aug 02 10:22:09 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:09] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
Aug 02 10:22:09 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:09] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
Aug 02 10:22:09 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:09] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
Aug 02 10:22:13 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:13] "GET / HTTP/1.1" 200 -
Aug 02 10:22:13 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:13] "GET / HTTP/1.1" 200 -
Aug 02 10:22:13 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:13] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
Aug 02 10:22:13 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:13] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
Aug 02 10:22:13 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:13] "GET /static/css/bootstrap-icons.css HTTP/1.1" 304 -
Aug 02 10:22:13 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:13] "GET /static/css/bootstrap-icons.css HTTP/1.1" 304 -
Aug 02 10:22:13 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:13] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
Aug 02 10:22:13 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:13] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
Aug 02 10:22:13 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:13] "GET /static/gallery/snapshot/pimage_preview_0.jpg HTTP/1.1" 304 -
Aug 02 10:22:13 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:13] "GET /static/gallery/snapshot/pimage_preview_0.jpg HTTP/1.1" 304 -
Aug 02 10:22:13 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:13] "GET /static/img/ov5647.jpg HTTP/1.1" 304 -
Aug 02 10:22:13 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:13] "GET /static/img/ov5647.jpg HTTP/1.1" 304 -
Aug 02 10:22:13 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:13] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
Aug 02 10:22:13 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:13] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
Aug 02 10:22:16 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:16] "GET / HTTP/1.1" 200 -
Aug 02 10:22:16 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:16] "GET / HTTP/1.1" 200 -
Aug 02 10:22:16 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:16] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
Aug 02 10:22:16 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:16] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
Aug 02 10:22:16 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:16] "GET /static/img/ov5647.jpg HTTP/1.1" 304 -
Aug 02 10:22:16 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:16] "GET /static/img/ov5647.jpg HTTP/1.1" 304 -
Aug 02 10:22:16 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:16] "GET /static/css/bootstrap-icons.css HTTP/1.1" 304 -
Aug 02 10:22:16 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:16] "GET /static/css/bootstrap-icons.css HTTP/1.1" 304 -
Aug 02 10:22:16 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:16] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
Aug 02 10:22:16 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:16] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
Aug 02 10:22:16 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:16] "GET /static/gallery/snapshot/pimage_preview_0.jpg HTTP/1.1" 304 -
Aug 02 10:22:16 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:16] "GET /static/gallery/snapshot/pimage_preview_0.jpg HTTP/1.1" 304 -
Aug 02 10:22:16 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:16] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
Aug 02 10:22:16 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:16] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
Aug 02 10:22:19 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:19] "GET /control_camera_0 HTTP/1.1" 200 -
Aug 02 10:22:19 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:19] "GET /control_camera_0 HTTP/1.1" 200 -
Aug 02 10:22:19 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:19] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
Aug 02 10:22:19 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:22:19] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
Aug 02 10:22:19 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:22:19] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
[... 128 lines cut ...]
Aug 02 10:25:18 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:25:18] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
Aug 02 10:26:33 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:26:33] "GET /control_camera_0 HTTP/1.1" 200 -
Aug 02 10:26:33 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:26:33] "GET /control_camera_0 HTTP/1.1" 200 -
Aug 02 10:26:33 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:26:33] "GET /static/css/bootstrap-icons.css HTTP/1.1" 304 -
Aug 02 10:26:33 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:26:33] "GET /static/css/bootstrap-icons.css HTTP/1.1" 304 -
Aug 02 10:26:33 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:26:33] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
Aug 02 10:26:33 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:26:33] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
Aug 02 10:26:33 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:26:33] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
Aug 02 10:26:33 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:26:33] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
Aug 02 10:26:33 kamera3 python[48510]: 172.17.1.98 - - [02/Aug/2024 10:26:33] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
Aug 02 10:26:33 kamera3 python[48510]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 10:26:33] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
Aug 02 10:46:23 kamera3 python[48510]: ----------------------------------------
Aug 02 10:46:23 kamera3 python[48510]: Exception occurred during processing of request from ('172.17.1.113', 45444)
Aug 02 10:46:23 kamera3 python[48510]: Traceback (most recent call last):
Aug 02 10:46:23 kamera3 python[48510]:   File "/usr/lib/python3.11/socketserver.py", line 317, in _handle_request_noblock
Aug 02 10:46:23 kamera3 python[48510]:     self.process_request(request, client_address)
[... cut ...] 48510]:   File "/usr/lib/python3.11/socketserver.py", line 705, in process_request
Aug 02 10:46:23 kamera3 python[48510]:     t.start()
Aug 02 10:46:23 kamera3 python[48510]:   File "/usr/lib/python3.11/threading.py", line 957, in start
Aug 02 10:46:23 kamera3 python[48510]:     _start_new_thread(self._bootstrap, ())
Aug 02 10:46:23 kamera3 python[48510]: RuntimeError: can't start new thread
Aug 02 10:46:23 kamera3 python[48510]: ----------------------------------------

Kind regards,
Aleksander

AleksanderKruecken, 2024-08-02:
Hi,

I have now installed Picamera2 webui on a fresh Raspberry pi 5 (8 GB memory).
The raspberry 5 has the fresh and latest RPI OS Bookworm installed and Raspberry Pi Camera module 3 connected to it.
Nothing else is installed on the Pi.
After some time the web UI doesn't show the stream in the camera feed tab, but the tabs of the GUI work.

sudo systemctl status picamera2-webui.service
● picamera2-webui.service - Picamera2 WebUI Server
     Loaded: loaded (/etc/systemd/system/picamera2-webui.service; enabled; preset: enabled)
     Active: active (running) since Fri 2024-08-02 13:08:04 CEST; 1h 16min ago
   Main PID: 912 (python)
      Tasks: 88 (limit: 9247)
        CPU: 2h 30min 4.224s
     CGroup: /system.slice/picamera2-webui.service
             └─912 /usr/bin/python /home/artes/picamera2-WebUI/app.py

 journalctl -u picamera2-webui.service -f -n 1000
Aug 02 14:17:19 kamera3 python[912]: INFO:picamera2.request:Saved <picamera2.request.Helpers object at 0x7fff9d7018d0> to file /home/artes/picamera2-WebUI/static/gallery/snapshot/pimage_snapshot_0.jpg.
Aug 02 14:17:19 kamera3 python[912]: picamera2.request INFO: Time taken for encode: 19.27227799933462 ms.
Aug 02 14:17:19 kamera3 python[912]: INFO:picamera2.request:Time taken for encode: 19.27227799933462 ms.
Aug 02 14:17:20 kamera3 python[912]: 172.17.1.116 - - [02/Aug/2024 14:17:20] "GET /snapshot_0 HTTP/1.1" 200 -
Aug 02 14:17:20 kamera3 python[912]: INFO:werkzeug:172.17.1.116 - - [02/Aug/2024 14:17:20] "GET /snapshot_0 HTTP/1.1" 200 -
Aug 02 14:17:22 kamera3 python[912]: 172.17.1.113 - - [02/Aug/2024 14:17:22] "GET /video_feed_0 HTTP/1.1" 200 -
Aug 02 14:17:22 kamera3 python[912]: INFO:werkzeug:172.17.1.113 - - [02/Aug/2024 14:17:22] "GET /video_feed_0 HTTP/1.1" 200 -
Aug 02 14:17:32 kamera3 python[912]: 172.17.1.113 - - [02/Aug/2024 14:17:32] "GET /video_feed_0 HTTP/1.1" 200 -
Aug 02 14:17:32 kamera3 python[912]: INFO:werkzeug:172.17.1.113 - - [02/Aug/2024 14:17:32] "GET /video_feed_0 HTTP/1.1" 200 -
Aug 02 14:17:41 kamera3 python[912]: picamera2.picamera2 DEBUG: Execute job: <picamera2.job.Job object at 0x7fff8c3c42d0>
Aug 02 14:17:41 kamera3 python[912]: DEBUG:picamera2.picamera2:Execute job: <picamera2.job.Job object at 0x7fff8c3c42d0>
Aug 02 14:17:41 kamera3 python[912]: picamera2.request INFO: Saved <picamera2.request.Helpers object at 0x7fff9d7018d0> to file /home/artes/picamera2-WebUI/static/gallery/snapshot/pimage_snapshot_0.jpg.
Aug 02 14:17:41 kamera3 python[912]: INFO:picamera2.request:Saved <picamera2.request.Helpers object at 0x7fff9d7018d0> to file /home/artes/picamera2-WebUI/static/gallery/snapshot/pimage_snapshot_0.jpg.
Aug 02 14:17:41 kamera3 python[912]: picamera2.request INFO: Time taken for encode: 10.726433999479923 ms.
Aug 02 14:17:41 kamera3 python[912]: INFO:picamera2.request:Time taken for encode: 10.726433999479923 ms.
Aug 02 14:17:42 kamera3 python[912]: 172.17.1.116 - - [02/Aug/2024 14:17:42] "GET /snapshot_0 HTTP/1.1" 200 -
Aug 02 14:17:42 kamera3 python[912]: INFO:werkzeug:172.17.1.116 - - [02/Aug/2024 14:17:42] "GET /snapshot_0 HTTP/1.1" 200 -
Aug 02 14:17:50 kamera3 python[912]: picamera2.picamera2 DEBUG: Execute job: <picamera2.job.Job object at 0x7fff8c3c5410>
Aug 02 14:17:50 kamera3 python[912]: DEBUG:picamera2.picamera2:Execute job: <picamera2.job.Job object at 0x7fff8c3c5410>
Aug 02 14:17:50 kamera3 python[912]: picamera2.request INFO: Saved <picamera2.request.Helpers object at 0x7fff9d7018d0> to file /home/artes/picamera2-WebUI/static/gallery/snapshot/pimage_snapshot_0.jpg.
Aug 02 14:17:50 kamera3 python[912]: INFO:picamera2.request:Saved <picamera2.request.Helpers object at 0x7fff9d7018d0> to file /home/artes/picamera2-WebUI/static/gallery/snapshot/pimage_snapshot_0.jpg.
Aug 02 14:17:50 kamera3 python[912]: picamera2.request INFO: Time taken for encode: 12.97550700019201 ms.
Aug 02 14:17:50 kamera3 python[912]: INFO:picamera2.request:Time taken for encode: 12.97550700019201 ms.
Aug 02 14:17:51 kamera3 python[912]: 172.17.1.116 - - [02/Aug/2024 14:17:51] "GET /snapshot_0 HTTP/1.1" 200 -
Aug 02 14:17:51 kamera3 python[912]: INFO:werkzeug:172.17.1.116 - - [02/Aug/2024 14:17:51] "GET /snapshot_0 HTTP/1.1" 200 -
Aug 02 14:18:17 kamera3 python[912]: 172.17.1.98 - - [02/Aug/2024 14:18:17] "GET /control_camera_0 HTTP/1.1" 200 -
Aug 02 14:18:17 kamera3 python[912]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 14:18:17] "GET /control_camera_0 HTTP/1.1" 200 -
Aug 02 14:18:17 kamera3 python[912]: 172.17.1.98 - - [02/Aug/2024 14:18:17] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
Aug 02 14:18:17 kamera3 python[912]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 14:18:17] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
Aug 02 14:18:17 kamera3 python[912]: 172.17.1.98 - - [02/Aug/2024 14:18:17] "GET /static/css/bootstrap-icons.css HTTP/1.1" 304 -
Aug 02 14:18:17 kamera3 python[912]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 14:18:17] "GET /static/css/bootstrap-icons.css HTTP/1.1" 304 -
Aug 02 14:18:17 kamera3 python[912]: 172.17.1.98 - - [02/Aug/2024 14:18:17] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
Aug 02 14:18:17 kamera3 python[912]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 14:18:17] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
Aug 02 14:18:17 kamera3 python[912]: 172.17.1.98 - - [02/Aug/2024 14:18:17] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
[... cut ...]
Aug 02 14:18:20 kamera3 python[912]: 172.17.1.98 - - [02/Aug/2024 14:18:20] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
Aug 02 14:18:20 kamera3 python[912]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 14:18:20] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
Aug 02 14:18:20 kamera3 python[912]: 172.17.1.98 - - [02/Aug/2024 14:18:20] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
Aug 02 14:18:20 kamera3 python[912]: INFO:werkzeug:172.17.1.98 - - [02/Aug/2024 14:18:20] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -

AleksanderKruecken, 2024-08-02:
If I restart the service the stream becomes available again in the web gui (both on RPI zero2 and RPI5):
sudo systemctl restart picamera2-webui.service

monkeymademe, 2024-08-02:
I can't immediately say what the issue is.... well I can but how to fix it is another matter. I know I am not handling threads well so I will give that another look.

But... PiZero even a 2w might experience issues especially with memory usage and limited cores so there is a degree of 'maybe the webui is not compatible'. I am not saying it wont work just that I have not run tests on that hardware so I can't say its stable.

I got a makerfaire in the coming weeks so I will look at it as soon as I can

monkeymademe, 2024-10-01:
I made some improvements in branch v1.0.5-pre which might resolve the issue

I have still seen some issues with home assistant itself so will keep this going for another look

monkeymademe, 2024-10-21:
I am going to close this for now as I think I might have resolved the issue at least to the best I can on my side.