monkeymademe/CamUI #80: Cannot retrive video feed with Pi 3B+ and HQ Camera

dariotarantini, 2024-08-25:
I cant get video feed when using HQ camera in mode 4056x3040. Web page refreshing never ends, seems there is something broken with backend. First two modes works fine.

Hardware :
Raspberry PI 3B+, HQ Camera. I also tested RPI zero 2 with same results. 
[image]
[image]

monkeymademe, 2024-08-28:
I see that there is an issue but unfortunately can't say what or how to fix it right now.. I will poke around, I am on it ;-P

monkeymademe, 2024-08-28:
Yeah seems there is a memory allocation issue when running on older pi's... If you start from the bottom and work your way up the resolutions the pi4 can do a little better but the issue persists when you get to the biggest resolution.

Unfortunately looks like a picamera2 issue and I am still investigating maybe... Maybe allowing changes to the buffers might help I am not sure for now I would recommend upgrading to a pi5 (the 2bg model has just released) or just stick to the lower resolution for the time being

I created a feature request to deal with explaining this in the frontend https://github.com/monkeymademe/picamera2-WebUI/issues/81

Sorry this is maybe not the answer you where looking for

dariotarantini, 2024-08-28:
Hi! Thank you actually, I own a rpi 5 so I can just try it there! i will let you know in 7/8 h. im doing a bit of astrophotography so higher resolutions are better for my case. I guess pi3/pi0v2 is just not enough. Also, is there any way to get "the feed" in a lower res and take pictures in full res? instead of sending everything in full resolution

monkeymademe, 2024-08-28:
I like the cut of your jib sir!

[image]

To answer your question... yes and no its on my todo... there is a way to view different feeds but it does not mean the higher resolution feed is not running just means you have a choice between a RAW feed a Main feed (the one you will see in the UI) and a lores feed (if enabled) its a tricky juggling act.

There is also a function for switching config so video config (which is what UI is using) has one set of settings and still config at the moment the UI is only using video and the still is saved from the feed. 'switch_mode_and_capture_image' function in picamera2 should switch but last I used that the feed did not recover so it was not used. Its because the feeds not connected to actual video config so when you switch mode the feed bombs out in an error because the still config is being used... if that makes sense.. or if I got that right!

monkeymademe, 2024-10-01:
Ahh good can I close this issue?

penlok, 2025-01-19:
Hi, I'm also trying to use RPi3B with HQ camera. I am facing the same memory allocation issue with the highest resolution 4056x3040. But in addition I can't get the feed to work at half resolution 2028x1520. However, in this case it is possible to capture the image. I don't know if this is a related issue or not but I don't get the memeory allocation error (in the journal)

This is what I get instead for this case from journalctl:
Jan 19 14:57:18 Camera kernel: bcm2835-codec bcm2835-codec: Failed setting ctrl 00990a67, ret -3
Jan 19 14:57:19 Camera kernel: bcm2835-codec bcm2835-codec: Failed setting ctrl 00990a67, ret -3
Jan 19 14:57:19 Camera kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -2

Another strange observation is that anytime I change the resolution (even with success for the feed), the app printout shows three equal configurations in sequence of the camera. Two of them also shows up in the journal as (similar to first two lines above):

Jan 19 14:41:29 Camera kernel: bcm2835-codec bcm2835-codec: Failed setting ctrl 00990a67, ret -3
Jan 19 14:41:30 Camera kernel: bcm2835-codec bcm2835-codec: Failed setting ctrl 00990a67, ret -3

and it seems that the first line in the journal occurs twice then - even when the feed works. For each of the three occurences, the app printout (actually a Picamera2 printout) says that buffers are allocated. If this is true, it could of course also be a cause of memory allocation problem.

Pen

closed 2025-04-10 by monkeymademe