monkeymademe/CamUI #84: Raspberry camera Module 3 NoIR Wide not detected in picamera2-webui

AleksanderKruecken, 2024-08-30:
**Describe the bug**
I have Raspberry pi zero2W and Raspberry pi camera module 3
I have installed picamera2-webui on Raspberry pi zero2w running RPI OS Bookworm 64bit lite, but can't see conected cameras in picamera2-webui GUI interface. The camera is connected correctly and I can make pictures and videos via libcamera-still and libcamera-vid commands.

**To Reproduce**
Steps to reproduce the behavior:
- I have installed Raspberry pi OS Bookworm 64bit lite on RPI Zero2W
-Updatee the package lists
sudo apt update
- Upgradee all the installed packages to their latest versions
sudo apt upgrade -y
- Installed Git
sudo apt install git -y
- Installed Flask
sudo apt install python3-flask -y
- Installed Pillow
sudo apt install python3-pillow -y
- Installed picamera2
sudo apt install python3-picamera2 -y
- Made sure that the camera is working by taking a picture and video and checked that everything was recorded
libcamera-still -o still-test.jpg
libcamera-vid -t 10000 -o test.h264
- Cloned the repository to my Raspberry Pi
git clone https://github.com/monkeymademe/picamera2-WebUI.git
- Entered the directory and run the app
cd picamera2-WebUI
python app.py
- Accessed the web interface through my browser:
[http://myIP:8080/](http://172.17.1.5:8080/)

**Expected behavior**
I expected to see the camera in picamera2-webui

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Windows 11 Pro OS
 - Chrome Browser 128.0.6613.113

picamera2-webui log:
artes@kamera2:~/picamera2-WebUI $ python app.py
[1:17:36.330456237] [11677]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+65-6ddd79b5
[1:17:36.476180511] [11680]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:17:36.481578264] [11680]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media1 and ISP device /dev/media2
[1:17:36.481789047] [11680]  INFO RPI pipeline_base.cpp:1104 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
/home/artes/picamera2-WebUI/static/camera_config
Detected Cameras:
[{'Model': 'imx708_wide_noir', 'Location': 2, 'Rotation': 180, 'Id': '/base/soc/i2c0mux/i2c@1/imx708@1a', 'Num': 0}]
Camera Info:
{'Model': 'imx708_wide_noir', 'Location': 2, 'Rotation': 180, 'Id': '/base/soc/i2c0mux/i2c@1/imx708@1a', 'Num': 0}
Current detected compatible Cameras:
{'cameras': []}
[image]
[image]
[image]
[image]
 * Serving Flask app 'app'
 * Debug mode: off
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on http://127.0.0.1:8080
 * Running on http://172.17.1.5:8080
Press CTRL+C to quit
10.176.27.4 - - [30/Aug/2024 11:09:49] "GET / HTTP/1.1" 200 -
10.176.27.4 - - [30/Aug/2024 11:09:49] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:09:49] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:09:49] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:09:49] "GET /static/css/bootstrap-icons.css HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:09:49] "GET /static/css/fonts/bootstrap-icons.woff2?dd67030699838ea613ee6dbda90effa6 HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:09:52] "GET / HTTP/1.1" 200 -
10.176.27.4 - - [30/Aug/2024 11:09:52] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:09:53] "GET /static/css/bootstrap-icons.css HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:09:54] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:09:54] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
Camera Data []
[]
10.176.27.4 - - [30/Aug/2024 11:10:08] "GET /image_gallery HTTP/1.1" 200 -
10.176.27.4 - - [30/Aug/2024 11:10:08] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:10:08] "GET /static/css/bootstrap-icons.css HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:10:09] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:10:09] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:10:10] "GET /about HTTP/1.1" 200 -
10.176.27.4 - - [30/Aug/2024 11:10:10] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:10:10] "GET /static/css/bootstrap-icons.css HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:10:10] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:10:10] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:17:56] "GET / HTTP/1.1" 200 -
10.176.27.4 - - [30/Aug/2024 11:17:56] "GET /static/css/bootstrap.min.css HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:17:56] "GET /static/css/bootstrap-icons.css HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:17:57] "GET /static/js/jquery-3.7.1.min.js HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:17:57] "GET /static/css/fonts/bootstrap-icons.woff2?dd67030699838ea613ee6dbda90effa6 HTTP/1.1" 304 -
10.176.27.4 - - [30/Aug/2024 11:17:57] "GET /static/js/bootstrap.bundle.min.js HTTP/1.1" 304 -

monkeymademe, 2024-08-30:
Haaa I don't have a noir Cam so I did not know the model name is different... 

I'll add it in later (sorry not at a system that I can update and test the code): If you do this, it 'should' fix the issue

On another note I should work on fixing unknown cameras something is a miss there

You can edit this file - camera-module-info.json

Insert this: 

        {
            "module_name": "Camera Module v3 Wide Angle NoIR",
            "sensor_model": "imx708_wide_noir",
            "manufacturer": "Sony",
            "megapixels": 11.9,
            "sensor_resolution": "4608 x 2592 pixels",
            "video_modes": "2304 × 1296p56, 2304 × 1296p30 HDR, 1536 × 864p120",
            "sensor_image_area": "6.45 x 3.63mm (7.4mm diagonal)",
            "pixel_size": "1.4 µm x 1.4 µm",
            "optical_size": "1/2.43",
            "focus": "Motorized",
            "dof": "Approx 5 cm to ∞",
            "focal_length": "2.75mm",
            "horizontal_fov": "102 degrees",
            "vertical_fov": "67 degrees",
            "f_stop": "F2.2",
            "max_exposure_time": "112",
            "lens_mount": "N/A",
            "got_noir": "Yes",
            "is_pi_cam": true
        },

monkeymademe, 2024-09-12:
This camera has been added to v1.0.4 which is also on main

closed 2024-09-12 by monkeymademe