monkeymademe/CamUI #87: Problem with Libcamera I guess

TimPGamer, 2024-09-10:
**Describe the bug**
I wanted to start PiCamera WebGUi as described in Readme.md, just didnt work. Workshop without Camera, but Not with. Idk why it stopped working with this new Version, but u think it has to do something with libcamera, cause thats Part of the error at the end of the Log. I dont need this immediately fixed, would just be nice to have fun again with this awesome “program”.

**To Reproduce**
Pi 4B Rev 1.2 4GB ram,
NON RPi-Cam*,
Libcamera Newest version,
PiCamera WebGUi newest version

*did work with both libcamera and PiCamera WebGUi in the past,
 why non RPi-Cam? Cause i wanted Night vis, with Infrared lights all in one.

**Expected behavior**
I expected it to run, just like the Last Version did about 2 moths ago.

**Screenshots**
No Screenshot, but a Long log i copied from console. See below

**Log**

-Start Log-

admin@raspberrypi:~ $ cd picamera2-WebUI/
admin@raspberrypi:~/picamera2-WebUI $ python app.py
[0:03:31.325411790] [2139]  INFO Camera camera_manager.cpp:316 libcamera v0.3.1+50-69a894c4
[0:03:31.355802562] [2142]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:03:31.358601947] [2142]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media2 and ISP device /dev/media0
[0:03:31.358727483] [2142]  INFO RPI pipeline_base.cpp:1125 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
/home/admin/picamera2-WebUI/static/camera_config

Detected Cameras:
[{'Model': 'ov5647', 'Location': 2, 'Rotation': 0, 'Id': '/base/soc/i2c0mux/i2c@1/ov5647@36', 'Num': 0}]

Camera Info:
{'Model': 'ov5647', 'Location': 2, 'Rotation': 0, 'Id': '/base/soc/i2c0mux/i2c@1/ov5647@36', 'Num': 0}

Camera config has changed since last boot - Adding new Camera
picamera2.picamera2 INFO: Initialization successful.
picamera2.picamera2 INFO: Camera now open.
picamera2.picamera2 DEBUG: <libcamera._libcamera.CameraManager object at 0x7fa69625f0>
picamera2.picamera2 DEBUG: Requesting configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': {'format': 'SGBRG10_CSI2P', 'size': (640, 480), 'stride': 800, 'framesize': 384000}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'sensor': {'bit_depth': 10, 'output_size': (640, 480)}, 'display': 'main', 'encode': 'main'}
[0:03:31.368744883] [2139]  INFO Camera camera.cpp:1191 configuring streams: (0) 640x480-XBGR8888 (1) 640x480-SGBRG10_CSI2P
[0:03:31.369191694] [2142]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
picamera2.picamera2 INFO: Configuration successful!
picamera2.picamera2 DEBUG: Final configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': {'format': 'SGBRG10_CSI2P', 'size': (640, 480), 'stride': 800, 'framesize': 384000}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'sensor': {'bit_depth': 10, 'output_size': (640, 480)}, 'display': 'main', 'encode': 'main'}
picamera2.picamera2 DEBUG: Streams: {'main': <libcamera._libcamera.Stream object at 0x7fa6974b30>, 'lores': None, 'raw': <libcamera._libcamera.Stream object at 0x7f8dafd870>}
picamera2 DEBUG: Allocated 4 buffers for stream 0 with fds [24, 27, 30, 33]
picamera2 DEBUG: Allocated 4 buffers for stream 1 with fds [36, 39, 42, 45]
picamera2.picamera2 DEBUG: Requesting configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': {'format': 'SGBRG10_CSI2P', 'size': (1296, 972), 'stride': 1632, 'framesize': 1586304}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'sensor': {'bit_depth': 10, 'output_size': (1296, 972)}, 'display': 'main', 'encode': 'main'}
[0:03:31.379945607] [2139]  INFO Camera camera.cpp:1191 configuring streams: (0) 640x480-XBGR8888 (1) 1296x972-SGBRG10_CSI2P
[0:03:31.380366993] [2142]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
picamera2.picamera2 INFO: Configuration successful!
picamera2.picamera2 DEBUG: Final configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': {'format': 'SGBRG10_CSI2P', 'size': (1296, 972), 'stride': 1632, 'framesize': 1586304}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'sensor': {'bit_depth': 10, 'output_size': (1296, 972)}, 'display': 'main', 'encode': 'main'}
picamera2.picamera2 DEBUG: Streams: {'main': <libcamera._libcamera.Stream object at 0x7fa6974b30>, 'lores': None, 'raw': <libcamera._libcamera.Stream object at 0x7f8dafd870>}
picamera2 DEBUG: Allocated 4 buffers for stream 0 with fds [24, 27, 30, 33]
picamera2 DEBUG: Allocated 4 buffers for stream 1 with fds [36, 39, 42, 45]
picamera2.picamera2 DEBUG: Requesting configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': {'format': 'SGBRG10_CSI2P', 'size': (1920, 1080), 'stride': 2400, 'framesize': 2592000}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'sensor': {'bit_depth': 10, 'output_size': (1920, 1080)}, 'display': 'main', 'encode': 'main'}
[0:03:31.393033816] [2139]  INFO Camera camera.cpp:1191 configuring streams: (0) 640x480-XBGR8888 (1) 1920x1080-SGBRG10_CSI2P
[0:03:31.393432387] [2142]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA
picamera2.picamera2 INFO: Configuration successful!
picamera2.picamera2 DEBUG: Final configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': {'format': 'SGBRG10_CSI2P', 'size': (1920, 1080), 'stride': 2400, 'framesize': 2592000}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'sensor': {'bit_depth': 10, 'output_size': (1920, 1080)}, 'display': 'main', 'encode': 'main'}
picamera2.picamera2 DEBUG: Streams: {'main': <libcamera._libcamera.Stream object at 0x7fa6974b30>, 'lores': None, 'raw': <libcamera._libcamera.Stream object at 0x7f8dafd870>}
picamera2 DEBUG: Allocated 4 buffers for stream 0 with fds [24, 27, 30, 33]
picamera2 DEBUG: Allocated 4 buffers for stream 1 with fds [36, 39, 42, 45]
picamera2.picamera2 DEBUG: Requesting configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': {'format': 'SGBRG10_CSI2P', 'size': (2592, 1944), 'stride': 3264, 'framesize': 6345216}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'sensor': {'bit_depth': 10, 'output_size': (2592, 1944)}, 'display': 'main', 'encode': 'main'}
[0:03:31.407330274] [2139]  INFO Camera camera.cpp:1191 configuring streams: (0) 640x480-XBGR8888 (1) 2592x1944-SGBRG10_CSI2P
[... cut ...]
python: ../include/libcamera/controls.h:167: T libcamera::ControlValue::get() const [with T = libcamera::Span<const libcamera::Rectangle>; typename std::enable_if<(libcamera::details::is_span<U>::value || std::is_same<std::__cxx11::basic_string<char>, typename std::remove_cv< <template-parameter-1-1> >::type>::value), std::nullptr_t>::type <anonymous> = nullptr]: Assertion `isArray_' failed.
Aborted

-End Log-

**Additional context**
Just tried to run After freshly installing. Wanted to take a Look at the changes i just recently saw.

monkeymademe, 2024-09-11:
2 things... which pi OS version are you running and maybe try:

`sudo apt upgrade`

I will try and have a look. Maybe there is an issue with a setting I recently added (but not sure how)

Line 168 of the app.py can you use this:

`if control in ['ScalerCrop', 'AfPause', 'FrameDurationLimits', 'NoiseReductionMode', 'AfMetering', 'ColourGains', 'StatsOutputEnable', 'AnalogueGain', 'AfWindows', 'AeFlickerPeriod', 'HdrMode', 'AfTrigger']:`

Then run and see if that improved basically it puts AnalogueGain back in the filter list so the setting is not sent to picamera/libcamera

monkeymademe, 2024-09-11:
If that works let me know

TimPGamer, 2024-09-11:
I’ll try asap, but not at home rn

TimPGamer, 2024-09-11:
Okay, I tried:

- Inserting AnalogueGain
- sudo apt update && sudo apt upgrade -y
- reinstalling everything

But sadly nothing worked.

TimPGamer, 2024-09-11:
I just ran update and upgrade again and its now at least upgrading something will tray again

monkeymademe, 2024-09-11:
Yeah there was another report so suspect something is updated in libcamera that's thrown the system I'll be looking at it tonight/tomorrow

TimPGamer, 2024-09-11:
Okay, thank u

TimPGamer, 2024-09-12:
Just the commit to v1.0.4 u think that works?

monkeymademe, 2024-09-12:
Fixed... libcamera either renamed 'Scalercrop' to 'Scalercrops' or added 'Scalercrops' 

But yeah with 'Scalercrops' not being in the default filter its then been set with something libcamera did not like.

https://github.com/monkeymademe/picamera2-WebUI/commit/e760c689cccc075f654f5643f5084aea6df5d56b Fixed on Main and v1.0.4 release

closed 2024-09-12 by monkeymademe