monkeymademe/CamUI #88: Picamera2-WebUI fails with controls.h . Assertion `isArray_' failed.

Steve8910, 2024-09-11:
**Describe the bug**
Fails with 

"python3: ../include/libcamera/controls.h:167: T libcamera::ControlValue::get() const [with T = libcamera::Span<const libcamera::Rectangle>; typename std::enable_if<(libcamera::details::is_span<U>::value || std::is_same<std::__cxx11::basic_string<char>, typename std::remove_cv< <template-parameter-1-1> >::type>::value), std::nullptr_t>::type <anonymous> = nullptr]: Assertion `isArray_' failed.
Aborted"

**To Reproduce**
Steps to reproduce the behavior:
1. Pi4 Clean Bookworm installed updated and upgraded.
2. Python3-flask Installed
3. Python3-picamera2 Installed.
4. git clone https://github.com/monkeymademe/picamera2-WebUI.git
5. ran python3 app.py in the picamera2-WebUI Folder.
6. Error seen

**Expected behavior**
No Error produced

monkeymademe, 2024-09-11:
I think maybe there is an issue from an update to libcamera or picamera2 I will check that out. Similar to https://github.com/monkeymademe/picamera2-WebUI/issues/87

I will check it out this evening. In the mean time can you tell me what camera you used

Steve8910, 2024-09-11:
I am using a PI HQ Camera 

Detected Cameras:
[{'Model': 'imx477', 'Location': 2, 'Rotation': 180, 'Id': '/base/soc/i2c0mux/i2c@1/imx477@1a', 'Num': 0}]

from the console output

C-Adam, 2024-09-11:
I am having the same issue.

Steps to the behavior:

Pi5 updated and upgraded.
git clone https://github.com/monkeymademe/picamera2-WebUI.git
ran python3 app.py in the picamera2-WebUI Folder.
Error seen

monkeymademe, 2024-09-12:
Fixed... libcamera either renamed 'Scalercrop' to 'Scalercrops' or added 'Scalercrops' 

But yeah with 'Scalercrops' not being in the default filter its then been set with something libcamera did not like.

https://github.com/monkeymademe/picamera2-WebUI/commit/e760c689cccc075f654f5643f5084aea6df5d56b

closed 2024-09-12 by monkeymademe