monkeymademe/CamUI #93: Wifi connection problems on Raspberry pi zero2w after installing picamera2-webui on them

AleksanderKruecken, 2024-09-12:
**Describe the bug**
I have a few Raspberry pi zero2w with Raspberry pi OS Bookworm lite (latest update and upgrade) and raspberry pi camera Module 3 NoIR Wide connected to them and all of them lose Wifi connection in a few hours after I install picamera2-WebUI on them. Otherwise they run just fine on wifi and they have nothing other installed on them beside git, python3-flask, python3-pillow, python3-picamera2. As I said all this installed packages run just fine and all the Raspberry pi zero2w don't lose wifi connection for weeks, but lost them in a few hours after I install and run picamera2-webUI on them.

I also have Raspberry pi 5 with RPI Camera module 3 and there are no problems.

**To Reproduce**
1.) Install fresh raspberry pi bookworm lite OS
2.) update&upgrade OS
3.) Install git, python3-flask, python3-pillow, python3-picamera2, picamera2-webUI
4.) Turn off power saving mode:
- Insert this line right below “# By default this script does nothing”
/etc/rc.local:
/sbin/iwconfig wlan0 power off
5.)Wait a few hours or even less and you can't connect to them via WiFi SSH.

**Expected behavior**
I would expect that wifi stays on.

**Desktop (please complete the following information):**
 - OS Raspberry bookworm lite
 - Version of picamera2-webui v1.0.3

journalctl:
kamera1 kernel: brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
kamera1 kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
kamera1 kernel: brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
kamera1 kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK

The log indicates issues with the communication between the Raspberry Pi's kernel and the Broadcom Wi-Fi/Bluetooth driver (brcmfmac) that's responsible for camera's wireless connection.

Sep 11 15:05:55 kamera1 kernel: brcmfmac: brcmf_sdio_isr: failed backplane access
Sep 11 15:06:24 kamera1 kernel: brcmfmac: brcmf_sdio_readframes: read 84 bytes from channel 2 failed: -110
Sep 11 15:06:24 kamera1 kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
Sep 11 15:06:25 kamera1 kernel: brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
Sep 11 15:06:25 kamera1 kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
Sep 11 15:10:05 kamera1 kernel: brcmfmac: brcmf_sdio_readframes: read 84 bytes from channel 2 failed: -110
Sep 11 15:10:05 kamera1 kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
Sep 11 15:15:22 kamera1 kernel: brcmfmac: brcmf_sdio_isr: failed backplane access
Sep 11 15:15:38 kamera1 kernel: brcmfmac: brcmf_sdio_isr: failed backplane access
Sep 11 15:15:41 kamera1 kernel: brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
Sep 11 15:15:41 kamera1 kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
Sep 11 15:15:43 kamera1 kernel: brcmfmac: brcmf_sdio_rxfail: count never zeroed: last 0xffff
Sep 11 15:15:46 kamera1 kernel: Unable to handle kernel NULL pointer dereference at virtual address 0000000000000038
Sep 11 15:15:46 kamera1 kernel: Mem abort info:
Sep 11 15:15:46 kamera1 kernel:   ESR = 0x0000000096000006
Sep 11 15:15:46 kamera1 kernel:   EC = 0x25: DABT (current EL), IL = 32 bits
Sep 11 15:15:46 kamera1 kernel:   SET = 0, FnV = 0
Sep 11 15:15:46 kamera1 kernel:   EA = 0, S1PTW = 0
Sep 11 15:15:46 kamera1 kernel:   FSC = 0x06: level 2 translation fault
Sep 11 15:15:46 kamera1 kernel: Data abort info:
Sep 11 15:15:46 kamera1 kernel:   ISV = 0, ISS = 0x00000006, ISS2 = 0x00000000
Sep 11 15:15:46 kamera1 kernel:   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
Sep 11 15:15:46 kamera1 kernel:   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
Sep 11 15:15:46 kamera1 kernel: user pgtable: 4k pages, 39-bit VAs, pgdp=0000000004d9c000
Sep 11 15:15:46 kamera1 kernel: [0000000000000038] pgd=0800000004efd003, p4d=0800000004efd003, pud=0800000004efd003, pmd=0000000000000000
Sep 11 15:15:46 kamera1 kernel: Internal error: Oops: 0000000096000006 [#1] PREEMPT SMP
Sep 11 15:15:46 kamera1 kernel: Modules linked in: cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep brcmfmac_wcc vc4 brcmfmac snd_soc_hdmi_codec drm_display_helper cec drm_dma_h>
Sep 11 15:15:46 kamera1 kernel: CPU: 3 PID: 76 Comm: kworker/u13:0 Tainted: G         C         6.6.31+rpt-rpi-v8 #1  Debian 1:6.6.31-1+rpt1
Sep 11 15:15:46 kamera1 kernel: Hardware name: Raspberry Pi Zero 2 W Rev 1.0 (DT)
Sep 11 15:15:46 kamera1 kernel: Workqueue: brcmf_wq/mmc1:0001:1 brcmf_sdio_dataworker [brcmfmac]
Sep 11 15:15:46 kamera1 kernel: pstate: 20000005 (nzCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Sep 11 15:15:46 kamera1 kernel: pc : bcm2835_mmc_transfer_dma+0x94/0x258
Sep 11 15:15:46 kamera1 kernel: lr : bcm2835_mmc_transfer_dma+0x84/0x258
Sep 11 15:15:46 kamera1 kernel: sp : ffffffc080dc3900
Sep 11 15:15:46 kamera1 kernel: x29: ffffffc080dc3900 x28: 0000000000002000 x27: 00000000000007e0
Sep 11 15:15:46 kamera1 kernel: x26: ffffff8002ae4e00 x25: ffffff8002f1f400 x24: ffffffc080dc3d50
Sep 11 15:15:46 kamera1 kernel: x23: ffffff8002f1f400 x22: 0000000000000001 x21: ffffff8002536480
Sep 11 15:15:46 kamera1 kernel: x20: 0000000000000005 x19: ffffff80054b0540 x18: 0000000000000000
Sep 11 15:15:46 kamera1 kernel: x17: ffffff97e4513000 x16: ffffffe8266305a8 x15: 0000007f931f40c0
Sep 11 15:15:46 kamera1 kernel: x14: 0000000000000001 x13: 0000000000000000 x12: ffffffe82693cbb0
Sep 11 15:15:46 kamera1 kernel: x11: ffffff80033de3c8 x10: 0000000000001a30 x9 : ffffffe825d382e0
Sep 11 15:15:46 kamera1 kernel: x8 : ffffff80054b0610 x7 : 0000000000000000 x6 : 000000003f000000
Sep 11 15:15:46 kamera1 kernel: x5 : ffffff80024a9b00 x4 : 00000000ffffffff x3 : 0000000000000000
Sep 11 15:15:46 kamera1 kernel: x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff8002502080
Sep 11 15:15:46 kamera1 kernel: Call trace:
Sep 11 15:15:46 kamera1 kernel:  bcm2835_mmc_transfer_dma+0x94/0x258
Sep 11 15:15:46 kamera1 kernel:  bcm2835_mmc_request+0xa4/0xc8
Sep 11 15:15:46 kamera1 kernel:  __mmc_start_request+0x74/0x1e0
Sep 11 15:15:46 kamera1 kernel:  mmc_start_request+0x8c/0xc0
Sep 11 15:15:46 kamera1 kernel:  mmc_wait_for_req+0x78/0x110
Sep 11 15:15:46 kamera1 kernel:  mmc_submit_one+0x98/0x130 [brcmfmac]
Sep 11 15:15:46 kamera1 kernel:  brcmf_sdiod_sglist_rw+0x230/0x530 [brcmfmac]
Sep 11 15:15:46 kamera1 kernel:  brcmf_sdiod_send_pkt+0x128/0x150 [brcmfmac]
Sep 11 15:15:46 kamera1 kernel:  brcmf_sdio_dataworker+0x1cc4/0x2990 [brcmfmac]
Sep 11 15:15:46 kamera1 kernel:  process_one_work+0x148/0x3b8
Sep 11 15:15:46 kamera1 kernel:  worker_thread+0x32c/0x450
Sep 11 15:15:46 kamera1 kernel:  kthread+0x11c/0x128
Sep 11 15:15:46 kamera1 kernel:  ret_from_fork+0x10/0x20
Sep 11 15:15:46 kamera1 kernel: Code: 2a0003f4 340009c0 f9407a61 f94002a0 (f9401c21)
Sep 11 15:15:46 kamera1 kernel: ---[ end trace 0000000000000000 ]---

monkeymademe, 2024-09-12:
Are they powering off and not reconnecting or are they still up running picamera2 webUI just without wifi?

I saw some funky business with throttling with over heating that might cause the wifi to drop... I'll have to see when I have the time to try and reproduce this.

AleksanderKruecken, 2024-09-13:
The zero2w RPI-s are still up running picamera2 webUI just without wifi.
I have tried with different RPI adaptors for RPI Zero2, but the problem remains. 
RPI Zero2 do heat up quite a lot. So much that one can not touch the CPU and barely the WiFi module.

Here is a sample of the log file when wifi stops working
Sep 12 14:28:33 kamera1 python[537]: 172.17.1.113 - - [12/Sep/2024 14:28:33] "GET /video_feed_0 HTTP/1.1" 200 -
Sep 12 14:35:02 kamera1 kernel: brcmfmac: brcmf_sdio_isr: failed backplane access
Sep 12 14:36:44 kamera1 kernel: brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
Sep 12 14:36:44 kamera1 kernel: brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Sep 12 14:36:44 kamera1 kernel: brcmfmac: brcmf_sdio_hdparse: HW header checksum error
Sep 12 14:36:44 kamera1 kernel: brcmfmac: brcmf_sdio_rxfail: terminate frame

monkeymademe, 2024-09-27:
While the wifi issue I can't pin down for sure why thats happening I have made a change in the following branch to see if the buffer is taking too much of the memory and its causing issues.

So I have the feed writing a frame to the buffer and then when the next frame comes it overwrites the last frame. 

This might improve the feed running long term and also improve performance in the flask side of things. (but also might lower the feed performance at higher resolutions... that might be the price to pay)

The fix is on https://github.com/monkeymademe/picamera2-WebUI/tree/v1.0.5-Pre branch if you would like to be my Guinea pig

monkeymademe, 2024-10-02:
I am closing this as the way I am dealing with the buffer seems to have improved the situation for others so I am hoping this is fixed as well. If not let me know and reopen the ticket

closed 2024-10-02 by monkeymademe