monkeymademe/CamUI #95: Error when running as a service

RickRoz, 2024-09-23:
When i am trying to check if the service is running with this command: 

sudo systemctl status picamera2-webui.service

I get the following response:

× picamera2-webui.service - Picamera2 WebUI Server
     Loaded: loaded (/etc/systemd/system/picamera2-webui.service; disabled; preset: enabled)
     Active: failed (Result: exit-code) since Mon 2024-09-23 19:32:14 CEST; 15s ago
   Duration: 52ms
    Process: 1910 ExecStart=/usr/bin/python /home/pi/picamera2-WebUI/app.py (code=exited, status=2)
   Main PID: 1910 (code=exited, status=2)
        CPU: 52ms

 picam systemd[1]: picamera2-webui.service: Scheduled restart job, restart counter is at 5.
 picam systemd[1]: Stopped picamera2-webui.service - Picamera2 WebUI Server.
 picam systemd[1]: picamera2-webui.service: Start request repeated too quickly.
 picam systemd[1]: picamera2-webui.service: Failed with result 'exit-code'.
 picam systemd[1]: Failed to start picamera2-webui.service - Picamera2 WebUI Server.

Hardware:
RPI 4

Software:
Raspberry pi OS (64bit) Bookworm 2024-07-04

monkeymademe, 2024-09-27:
What happens when you run the app in terminal (as in not as a service)

Also what is the username you login with on the pi... Mine is pi you might have something different

RickRoz, 2024-09-27:
> What happens when you run the app in terminal (as in not as a service)
> 
> Also what is the username you login with on the pi... Mine is pi you might have something different

the username was the problem. Thanks for the help!, got a new problem though and that the camera feed is not showing after some time. it stays blank until i reboot it.

monkeymademe, 2024-09-27:
yeah there has been some reports... I am building 2 things to try and fix it... The ability to change streaming encoders and stop and start the stream at will.

monkeymademe, 2024-09-27:
For the sake of the original issue working now though I will close this ticket

closed 2024-09-27 by monkeymademe