monkeymademe/CamUI #100: Problem to restart the WebUI with certain resolution settings

HessuHopo840, 2024-10-09:
I installed and tested the WebUI succesfully. Thanks for the project, it has significantly helped me.

Description of the problem: I launched WebUI first time and I saved a certain settings (AEC/AGC and AWB on). I also changed the resolution higher. The WebUI worked fine. However, when I restarted the interface. The interface returns error "[FileNotFoundError: [Errno 2] No such file or directory]" but there was no specific filename in the error. 

Temporal solution: There is a file refering to the settings profile that I saved  "test.json" for example in my case. I removed the file from path

`/picamera2-WebUI/static/camera_config/`

and after that the WebUI can be launched normally. A profile with just bare AEC/AGC and AWB on settings works fine.  Raspberry Pi Zero 2W is my board and the camera is the official V2.1.

monkeymademe, 2024-10-18:
I'll take a look thanks, not sure whats going on there....yet

monkeymademe, 2024-10-25:
I took a look and could not reproduce the issue. I did however find a small bug so I fixed that. Maybe its related I am not sure. 

If this happens again or you have more steps to reproduce then please reopen the ticket 

Also on a side note v1.0.5 will be released shortly which might improve things generally

closed 2024-10-25 by monkeymademe