monkeymademe/CamUI #101: Support tuning files for NoIR/greyworld

chris-y, 2024-10-15:
**Is your feature request related to a problem? Please describe.**
NoIR cameras show slightly purple.  This used to be resolved with a white balance mode called "greyworld", but now appears to need the use of tuning files.

**Describe the solution you'd like**
An option in the GUI to specify a tuning file.
The tuning files are at:
/usr/share/libcamera/ipa/rpi/vc4/

My camera is detected as `Detected camera 0: imx219`, so I need to use the imx219_noir.json tuning file.

I'm not sure if it is possible to autodetect the correct tuning file to use.  Maybe a "NoIR" checkbox could help here.

**Describe alternatives you've considered**
I tried adding awb_auto_is_greyworld=1 to config.txt but it looks like that no longer works.

**Additional context**
This might provide some useful information: https://github.com/raspberrypi/picamera2/issues/735

monkeymademe, 2024-10-18:
I am going to look into that but in the next version v1.0.16

TimPGamer, 2025-01-03:
I just figured out how to use Libcamera with the Tuning files if that Helps. In /usr/share/libcamera/ipa/rpi/vc4/sensor(_noir).json. The command to use is for me libcamera-still -o test.jpg —tuning-file /usr/share/libcamera/ipa/rpi/vc4/ov5647_noir.json to take still images. The tuning files in libcamera make it so that the photos take Look better. For more Infos Look at the following link. The topic is on how to make it better, but u can See a Great Exempel There. 
https://forums.raspberrypi.com/viewtopic.php?t=347194#p2079789

Nama, 2025-01-04:
Autodetecting would be fine, we need to be able to choose a tuning file ourselfs, since there are more files than just for NoIR.

You can read camera model with `picam2.camera_properties['Model']`

TimPGamer, 2025-01-04:
this says autodetecting doesnt work for V1 and V2. also maybe some others. https://forums.raspberrypi.com/viewtopic.php?t=347194#p2079838
But maybe thats wrong imma test that rq, cause i have a V1

TimPGamer, 2025-01-04:
weird 
[image]

TimPGamer, 2025-01-04:
so i imported picamera2 and just in case there is a module named picam2 i tried that too, but u see failed. so i tried the command. Doesnt exist.

TimPGamer, 2025-01-04:
holdup. Just looked in the code, cause i thought that iu might find ansers there and voila i think i need to import Picamera2 from picamera2 with the P being capital for a command to work that was used in the web app. so lets see if it works that way.

monkeymademe, 2025-01-04:
At the risk of stating what you aready discovered yes v1 and v2 cams can't tell the difference between the non and noir versions. V3 has a postfix on the sensor model 'imx708_noir' for example but v1 and v2 don't... So yeah when implementing this feature I would assume that you could try to 'upload' the file and point to this but then either picamera is ok with it and carries on or throws an error... I am not sure. I have the cameras to test it will slot it into the roadmap

Nama, 2025-01-04:
@TimPGamer hf ;)
https://github.com/raspberrypi/picamera2/tree/main/examples

B4875, 2025-05-25:
> [@TimPGamer](https://github.com/TimPGamer) hf ;) https://github.com/raspberrypi/picamera2/tree/main/examples

thx for useful link!! (I'l add example for jpeg)
I saw the same problem on 
https://forums.raspberrypi.com/viewtopic.php?t=285519, but that did not work for me...

```python
#!/usr/bin/python3

# Capture a JPEG while still running in the preview mode. When you
# capture to a file, the return value is the metadata for that image.

import datetime
import time

from picamera2 import Picamera2, Preview
from libcamera import Transform

tuning = Picamera2.load_tuning_file("imx219_noir.json")

algo = Picamera2.find_tuning_algo(tuning, "rpi.agc")
if "channels" in algo:
    algo["channels"][0]["exposure_modes"]["normal"] = {"shutter": [100, 66666], "gain": [1.0, 8.0]}
else:
    algo["exposure_modes"]["normal"] = {"shutter": [100, 66666], "gain": [1.0, 8.0]}

picam2 = Picamera2(tuning=tuning)

preview_config = picam2.create_preview_configuration(main={"size": (1640, 1232)}, transform = Transform(hflip = True, vflip = True))

picam2.configure(preview_config)

picam2.start()
#time.sleep(2)

metadata = picam2.capture_file("/home/test1.jpg")
print(metadata)

picam2.close()
```

TimPGamer, 2025-08-13:
I am not sure, if its possible like that, but maybe add a section in the cam controls maybe named Tuning Files, Where at the top you have a switch to activate it, then under it, a dropdown selection, to select from all tuning files, which will be greyed out, cause python can detect the sensor (I think). By default it will be the file for the cams with ir filter (only for V1, V2, any other that don’t differentiate IR and NoIR) so under the dropdown or besides it would be a check mark, which says “manual override” or something, so you can select another tuning files, for example if you have a NoIR cam or just wanna play around. This is just an idea for how to implement it. If i made an error or its already planned otherwise, then do it differently, i just thought id give some inspiration or something.