monkeymademe/CamUI #102: Is there Support for the INNO-MAKER CAM-OV9281RAW-V2 ?

EricLyons, 2024-10-17:
I have a INNO-MAKER CAM-MIPIOV9281-V2 module on board OmniVision’s OV9281 that is a global shutter camera. I'm currently running the built in bookworms drivers and not the InnoMaker Driver. The app detects the camera but doesn't see it has a compatible camera. The camera is working using the built in libcamera tools. Is there a way I can get this camera to work with this app? Thanks 

Detected Cameras:
[{'Model': 'ov9281', 'Location': 2, 'Rotation': 0, 'Id': '/base/soc/i2c0mux/i2c@1/ov9281@60', 'Num': 0}]

Camera Info:
{'Model': 'ov9281', 'Location': 2, 'Rotation': 0, 'Id': '/base/soc/i2c0mux/i2c@1/ov9281@60', 'Num': 0}

Current detected compatible Cameras:
{'cameras': []}

monkeymademe, 2024-10-18:
Actually you poked out a pretty sizable bug. 

To explain the system checks if the camera you have connected was connected before. Then its compared to the list of 'compatible cameras' which are more or less pi camera models at the moment. If the camera is not on the list and never been seen before the program 'should' build a default settings template based off of what picamera2 says it should be. 

The bug is that the last step is skipped if the camera is totally unknown to the system

A really stupid quick fix is to add the camera to the camera-module-info.json file.. What is the most important is the sensor_model... that should be ov9281 everything else is not important

Otherwise if you can give me a few days I can see if I can improve the camera detection and also add your camera in when I have more information.

EricLyons, 2024-10-18:
Thanks. I'll give that a shot.

monkeymademe, 2024-10-19:
So I have added the commit https://github.com/monkeymademe/picamera2-WebUI/commit/86ade9c4d2da963d95a7e3b2f09e025b7a466896 which 'should' fix the unknown could you give that a look to make sure it works

Also its possible that while picamera2 detects the camera it might not work due to camera limitations but that might be another ticket.

closed 2024-10-19 by monkeymademe

EricLyons, 2024-10-20:
It detected the camera and now it works from the UI.  Thanks