monkeymademe/CamUI #109: pip Package and/or deb Package

phistep, 2024-12-01:
The manual installation seems cumbersome and compicated for novice linux users. The way of running the server seems unpythonic. Setting up the systemd service has to be done manually and from reading the README.

Restructure the project root to match the [Python Packaging Authority recommended project layout][pyproject-layout]. Then use [`pyproject.toml`][pyproj.toml] to build a wheel package. Have it secify the server as entrypoint, so that it can be installed via pipx and run directly from the shell. Distribute these pre-built packages via PyPI and/or [Github Release Artifacts][gh-release] and use [Github Actions][gh-actions] to automate the build process on the CI.

Then the installation would be as simple as
```sh
pip3 install https://github.com/monkeymademe/picamera2-WebUI.git
# or 
pip3 install picamera2-WebUI
# or 
pipx install picamera2-WebUI

# then run
picamera2-webui
```

Also create a [Debian `.deb` package][raspbian-deb[ that can be installed via `apt-get`. This can then also pull-in any platform-specific system packages and ease the installation, as well as install systemd service files and run post-installation commands like starting the server. Distribute it via [Github Release Artifacts][gh-release] use [Github Actions][gh-actions] to automate the build process on the CI or even try to [get it upstreamed][raspbian-submit] into the [Rasbian repositories][raspbian-repo]. 

Then the installation would be as simple as
```sh
wget https://github.com/monkeymademe/picamera2-WebUI/releases/download/v1.0.0/picamera2-WebUI.deb
dpkg -i picamera2-WebUI.deb
# or 
apt-get install python3-picamera2-WebUI

# server started automatically
```

I am happy to do all this and open an PR, since I have done similar setups before, if it will get merged. Would you be interested in this?

[pyproject-layout]: https://packaging.python.org/en/latest/discussions/src-layout-vs-flat-layout/
[pyproject.toml]: https://packaging.python.org/en/latest/guides/writing-pyproject-toml/
[gh-release]: https://docs.github.com/en/repositories/releasing-projects-on-github/about-releases
[gh-actions]: https://docs.github.com/en/actions/about-github-actions/understanding-github-actions
[raspbian-deb]: https://forums.debian.net//viewtopic.php?f=16&t=38976
[raspbian-submit]: https://raspberrypi.stackexchange.com/a/37300
[raspbian-repo]: https://www.raspbian.org/RaspbianRepository

monkeymademe, 2025-04-10:
@phistep if you are still game to help out with this I would welcome it? I am not doing well with getting up pip/deb/docker and could use the help and explanation how

Mraanderson, 2025-11-07:
I think a deb installer would be great when the refinements are done and you are happy with a stable release. 

I used a launcher.sh file in a similar project to install or remove a system d service that ran the app.py at boot. If that is a suggestion that appeals feel free to have a root through it. The project is a bit up in the air.

https://github.com/Mraanderson/sstv-groundstation/blob/main/launcher.sh