monkeymademe/CamUI #110: ProcessLookupError: [Errno 3] No such process

chris-y, 2024-12-04:
**Describe the bug**
picamera2-WebUI not starting up

**To Reproduce**
Steps to reproduce the behavior:
1. Start the service 🙂

**Expected behavior**
Service starts and web page available

**Additional context**

I'm not sure when this started - my Pi crashed a while back and I only just got round to rebooting it.

I've updated to latest Git, although it looked like it was pretty much up to date anyway.

Log:
```
picamera2.picamera2 INFO: Camera started

Started Stream with encoder: MJPEGEncoder 

Started Stream with encoder: MJPEGEncoder 

Controls set successfully.

Live Config:
{'controls': {'Sharpness': 1.0, 'AeEnable': True, 'Contrast': 1.0, 'AnalogueGain': 3.9, 'AwbEnable': True, 'Saturation': 1.0, 'AeFlickerMode': 0, 'ExposureValue': 0.0, 'CnnEnableInputTensor': False, 'AeMeteringMode': 0, 'AwbMode': 0, 'AeConstraintMode': 0, 'Brightness': 0.0, 'ExposureTime': 0, 'AeExposureMode': 0}, 'rotation': {'hflip': 0, 'vflip': 0}, 'sensor-mode': 0, 'capture-settings': {'Resize': False, 'makeRaw': False, 'Resolution': 1, 'Encoder': 'MJPEGEncoder'}, 'GPIO': {'enableGPIO': False, 'button': 0, 'led': 0}, 'Model': 'imx219'}

Sensor Mode:
0

Traceback (most recent call last):
  File "/home/chris/picamera2-WebUI/app.py", line 528, in <module>
    camera_obj.start_streaming()
  File "/home/chris/picamera2-WebUI/app.py", line 239, in start_streaming
    self.camera.start_recording(MJPEGEncoder(), output=FileOutput(self.output))
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1815, in start_recording
    self.start_encoder(encoder, output, pts=pts, quality=quality, name=name)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1755, in start_encoder
    _encoder.start(quality=quality)
  File "/usr/lib/python3/dist-packages/picamera2/encoders/encoder.py", line 262, in start
    self._start()
  File "/usr/lib/python3/dist-packages/picamera2/encoders/mjpeg_encoder.py", line 43, in _start
    super()._start()
  File "/usr/lib/python3/dist-packages/picamera2/encoders/v4l2_encoder.py", line 149, in _start
    fcntl.ioctl(self.vd, VIDIOC_STREAMON, typev)
ProcessLookupError: [Errno 3] No such process
picamera2.picamera2 INFO: Camera stopped
picamera2.picamera2 INFO: Camera closed successfully.
```

chris-y, 2025-02-03:
I deleted the config files and then it started up. 
I thought I'd gone back and updated this, sorry!

monkeymademe, 2025-04-10:
This 'should' be resolved in v2.0.0 if not reopen this ticket

closed 2025-04-10 by monkeymademe