monkeymademe/CamUI #112: [Question] I made it run on the Model 2B, are you interested in the changes ?

szpeter80, 2024-12-06:
First, thanks for this awesome software !

I made it run on the Model 2B, are you interested in the changes ? My use case was to get a headless IP camera from the old hardware i had lying around. The code can be found here:

https://github.com/szpeter80/picamera2-WebUI/tree/szpeter-dev

 Some key points in random order:

- Used the "Raspberry Pi OS Lite" image and created a script to install all OS dependency packages
- Created a script to create and enable the systemd unit, as far as i see it works fine. By merging, help about making it run from systemd can be reduced and less error possiblity for the not so experienced.
- I found a 'turbo' version of  the libjpeg package. idk if this does the magic, but the Model2B with NoIR (probably v1) it can do 1296x972 (no crop in this mode) and stream it over a cheap wifi plug (wpa2), python process is about 85-90% (so nearly use up 1 from the 4 core), rest of the system uses less than 10% of a core, and from the 1G RAM with swap disabled, ~230MB used.  So it seems that this is useful software even on a such old board.
- Created a script to create a .venv as such to include the system packages and i made a requirements.txt and the packages specified gets installed to the venv as well
- Created a startup script which looks for the .venv, and if it is found it activates, then deactivate after the app finishes

If you are interested in some or all of the above changes please let me know. I can do proper, per-feature PR. 

with best regards,
Peter

Nama, 2024-12-26:
I don't understand. This just works on the 2B v1.1. No changes.

I also used picamera2 with my own scripts on 1B and zero W, so might even work on these, if the overhead is not too much with the webserver.

szpeter80, 2024-12-27:
Sorry, probably bad wording... it is rather 2 part, a statement of the project is usable on the model 2b (it is listed as a unsure on the compat list), and a real question, that are my changes (addititons of scripts etc) anything of interest for this project?

 I'm asking because the stuff "worksforme" and probably my first and last raspicam project, so i wont put too much effort to structure the changes if they are not relevant for other people's use case.
 
 I hope i was able to better phrase my intent this time.

monkeymademe, 2025-01-04:
I am interested in your changes thanks for that

closed 2025-04-10 by monkeymademe