monkeymademe/CamUI #114: Add docs for installation deps on lite

Routhinator, 2024-12-30:
There's mention of the deps, but we just need a one liner in the install directions:

```
apt-get install -y libcap-dev python3-libcamera
```

The project needs a requirements.txt for use with venvs, ideally. The following need to be installed in the venv:
```
pillow
gpiozero
picamera2
flask
```

Additionally the venv must either be created with `--system-site-packages` or the `<venv_dir>/pyvenv.cfg` must be modified so that `include-system-site-packages = true`

Routhinator, 2024-12-30:
Ok, I've verified these instructions through a few clean iterations now. If you like I can throw up a PR, I don't see a CONTRIB.md so I'm not sure if you are wanting PRs on this yet.

monkeymademe, 2025-01-04:
I am new to git (well not that new but still a noob) I did not know a contrib.md is a thing... PR away. I will look into the contrib file and make an update