monkeymademe/CamUI #116: Max exposure time changing

BlueLan0421, 2025-02-13:
I am currently using the picamera2 library with the picamera2-WebUI interface on my Raspberry Pi, and I have encountered a limitation where the maximum exposure time is capped at 66,666 microseconds (66.666 ms).

I would like to inquire if it is possible to extend this exposure time, either through a configuration change, firmware update, or by modifying the software settings. If there is a way to achieve this, I would greatly appreciate any guidance or instructions on how to proceed.

Thank you for your time and support. I look forward to your response.

monkeymademe, 2025-02-15:
I am fairly sure it's a limitation to the camera and set by picamera2 so I don't think there is much that can be done.

I found a great explanation why it's like that at this forum post - https://forums.raspberrypi.com/viewtopic.php?t=76242

But!.... slight hope might be the Picamera model GS (global shutter) since it does not a rolling shutter it might have a longer exposure setting. I'll plug mine in and see if it does later

BlueLan0421, 2025-02-17:
Based on the Raspberry Pi Camera specifications, the maximum supported exposure time is 11 seconds. I was able to capture a long-exposure image using the following Python script:

```
from picamera2 import Picamera2
import time

picam2 = Picamera2()
config = picam2.create_still_configuration()
picam2.configure(config)

picam2.set_controls({"ExposureTime": 2000000})

picam2.start()
time.sleep(12)
picam2.capture_file("long_exposure.jpg")
picam2.stop()
```
I suspect that the preview mode might be limiting the exposure time. If that's the case, would disabling the preview allow for longer exposures in the web interface?

Have you tested this on the Global Shutter (GS) Camera yet? It would be interesting to see if it behaves differently.

monkeymademe, 2025-02-17:
Hmmm I get the camera limits from picamera2 itself

a command like this -  print(picam2.camera_controls) would output the following

`{'HdrMode': (0, 4, 0), 'Contrast': (0.0, 32.0, 1.0), 'Brightness': (-1.0, 1.0, 0.0), 'ColourGains': (0.0, 32.0, None), 'AeFlickerPeriod': (100, 1000000, None), 'AwbMode': (0, 7, 0), 'AeFlickerMode': (0, 1, 0), 'AeExposureMode': (0, 3, 0), 'Sharpness': (0.0, 16.0, 1.0), 'AwbEnable': (False, True, None), 'Saturation': (0.0, 32.0, 1.0), 'StatsOutputEnable': (False, True, False), 'FrameDurationLimits': (33333, 120000, None), 'ScalerCrops': ((0, 0, 0, 0), (65535, 65535, 65535, 65535), (0, 0, 0, 0)), 'AeEnable': (False, True, None), 'ExposureTime': (0, 66666, None), 'AeMeteringMode': (0, 3, 0), 'NoiseReductionMode': (0, 4, 0), 'AnalogueGain': (1.0, 16.0, None), 'ScalerCrop': ((0, 0, 0, 0), (65535, 65535, 65535, 65535), (0, 0, 0, 0)), 'AeConstraintMode': (0, 3, 0), 'ExposureValue': (-8.0, 8.0, 0.0), 'CnnEnableInputTensor': (False, True, False)}`

Basically its 3 values Min, Max, Default - 'ExposureTime': (0, 66666, None) is where the 66666 is coming from.

Now it begs the question is that max value just place holder and the time can be longer? I need to run some tests.

monkeymademe, 2025-04-10:
Ah this is the ticket that spawned the whole reworking of the webUI

v2.0.0 should have fixed the issue ... enjoy!

closed 2025-04-10 by monkeymademe