monkeymademe/CamUI #119: Camera crashes on next reboot when config is saved

Zev-se, 2025-03-19:
**Describe the bug**
The project works fine as long as no configuration is saved. However next time I reboot (or restart the process) the script fails.

**To Reproduce**
Steps to reproduce the behavior:
1. Install raspberry Pi with 32-bit OS
2. apt-get update && apt-get upgrade && apt install -y python3-picamera2
3. git clone https://github.com/monkeymademe/picamera2-WebUI && cd picamera2-WebUI && python3 app.py
4. Go to webpage and change whatever settings and save
5. back in terminal ctrl+c and then python3 app.py

**Expected behavior**
Webserver should start again without issue and load saved settings.

**Screenshots**
```bash
Traceback (most recent call last):
  File "/home/zev/picamera2-WebUI/app.py", line 530, in <module>
    camera_obj.start_streaming()
  File "/home/zev/picamera2-WebUI/app.py", line 241, in start_streaming
    self.camera.start_recording(MJPEGEncoder(), output=FileOutput(self.output))
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1826, in start_recording
    self.start_encoder(encoder, output, pts=pts, quality=quality, name=name)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1766, in start_encoder
    _encoder.start(quality=quality)
  File "/usr/lib/python3/dist-packages/picamera2/encoders/encoder.py", line 281, in start
    self._start()
  File "/usr/lib/python3/dist-packages/picamera2/encoders/mjpeg_encoder.py", line 43, in _start
    super()._start()
  File "/usr/lib/python3/dist-packages/picamera2/encoders/v4l2_encoder.py", line 149, in _start
    fcntl.ioctl(self.vd, VIDIOC_STREAMON, typev)
FileNotFoundError: [Errno 2] No such file or directory
picamera2.picamera2 INFO: Camera stopped
picamera2.picamera2 INFO: Camera closed successfully.
```

**Desktop (please complete the following information):**
N/A

**Smartphone (please complete the following information):**
N/A

**Additional context**
This is tested on a Raspberry Pi 2B and a Raspberry pi zero v1. I do know these raspberry pi:s aren't tested but I'm not sure this has to do with the hardware.

monkeymademe, 2025-03-22:
Its got nothing to do with hardware but I am still unsure why you are getting this issue.

BUT... 

I totally rewrote all the code to improve how the system deals with config and how they are loading. 

At the moment this lives on another repo - https://github.com/monkeymademe/picamera2-WebUI-beta

you are welcome to try that out and see if it works better for you. I will be merging the beta with this repo when I think its ready (which will be soon!)

Keep an eye on the discussions for news when the merge will happen

Zev-se, 2025-03-29:
Nice improvements on the UI! I've tried it and I've had a few issues, some are my fault, some I can't figure out. I did try to just change the path in services but then the default config and the /static/saved-profiles gets owned by root. Thus running the service as a regular user from the directory first is a good idea. One can change them with chown of course. However, it still noes not work as a service as I can't run it through the complete path, but I can run them from the directory which is very weird.

```
# Through full path
zev@cam:~ $ /usr/bin/python3 /home/zev/picamera2-WebUI-beta/app.py
[0:03:18.686353000] [1123]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679
[0:03:19.218794000] [1217]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:03:19.257726000] [1217]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media2 and ISP device /dev/media0
[0:03:19.260988000] [1217]  INFO RPI pipeline_base.cpp:1121 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'

Initialize picamera2 - Cameras Found:
[{'Model': 'ov5647', 'Location': 2, 'Rotation': 0, 'Id': '/base/soc/i2c0mux/i2c@1/ov5647@36', 'Num': 0}]

Connected camera model 'ov5647' is found in the camera-module-info.json and is a Pi Camera.

[{'Num': 0, 'Model': 'ov5647', 'Is_Pi_Cam': True, 'Has_Config': False, 'Config_Location': 'default_ov5647.json'}]

 
[0:03:19.467166000] [1123]  INFO Camera camera.cpp:1202 configuring streams: (0) 640x480-XBGR8888 (1) 640x480-SGBRG10_CSI2P
[0:03:19.475925000] [1217]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
[0:03:19.632250000] [1123]  INFO Camera camera.cpp:1202 configuring streams: (0) 640x480-XBGR8888 (1) 1296x972-SGBRG10_CSI2P
[0:03:19.634449000] [1217]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
[0:03:19.784553000] [1123]  INFO Camera camera.cpp:1202 configuring streams: (0) 640x480-XBGR8888 (1) 1920x1080-SGBRG10_CSI2P
[0:03:19.790578000] [1217]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA
[0:03:19.937080000] [1123]  INFO Camera camera.cpp:1202 configuring streams: (0) 640x480-XBGR8888 (1) 2592x1944-SGBRG10_CSI2P
[0:03:19.940161000] [1217]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 2592x1944-SGBRG10_1X10 - Selected unicam format: 2592x1944-pGAA
Applied Orientation - hflip: False vflip: False
[0:03:20.131303000] [1123]  INFO Camera camera.cpp:1202 configuring streams: (0) 2592x1944-BGR888 (1) 2592x1944-SGBRG10_CSI2P
[0:03:20.135714000] [1217]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 2592x1944-SGBRG10_1X10 - Selected unicam format: 2592x1944-pGAA
Applied Orientation - hflip: False vflip: False
[0:03:20.266092000] [1123]  INFO Camera camera.cpp:1202 configuring streams: (0) 1280x720-XBGR8888 (1) 1920x1080-SGBRG10_CSI2P
[0:03:20.272818000] [1217]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA
Applied Orientation - hflip: False vflip: False
[0:03:20.648858000] [1221]  INFO Camera camera.cpp:1202 configuring streams: (0) 2592x1944-BGR888 (1) 640x480-SGBRG10_CSI2P
[0:03:20.680414000] [1217]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
Applied Orientation - hflip: False vflip: False
[0:03:20.811884000] [1221]  INFO Camera camera.cpp:1202 configuring streams: (0) 1280x720-XBGR8888 (1) 640x480-SGBRG10_CSI2P
[0:03:20.816504000] [1217]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
TESTING: {'Brightness': (-1.0, 1.0, 0.0), 'AnalogueGain': (1.0, 63.9375, 1.0), 'AwbMode': (0, 7, 0), 'AeMeteringMode': (0, 3, 0), 'ExposureTime': (134, 4879289, 20000), 'AeEnable': (False, True, None), 'ScalerCrop': ((16, 0, 256, 256), (16, 0, 2560, 1920), (16, 240, 2560, 1440)), 'FrameDurationLimits': (16971, 4879899, 33333), 'SyncFrames': (1, 1000000, 100), 'SyncMode': (0, 2, 0), 'StatsOutputEnable': (False, True, False), 'Saturation': (0.0, 32.0, 1.0), 'Sharpness': (0.0, 16.0, 1.0), 'Contrast': (0.0, 32.0, 1.0), 'NoiseReductionMode': (0, 4, 0), 'ColourGains': (0.0, 32.0, None), 'AeFlickerMode': (0, 1, 0), 'AeExposureMode': (0, 3, 0), 'ExposureValue': (-8.0, 8.0, 0.0), 'AwbEnable': (False, True, None), 'HdrMode': (0, 4, 0), 'AeConstraintMode': (0, 3, 0), 'AeFlickerPeriod': (100, 1000000, None), 'ColourTemperature': (100, 100000, None), 'CnnEnableInputTensor': (False, True, False)}
Traceback (most recent call last):
  File "/home/zev/picamera2-WebUI-beta/app.py", line 784, in <module>
    camera_obj = CameraObject(connected_camera)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zev/picamera2-WebUI-beta/app.py", line 175, in __init__
    self.live_controls = self.initialize_controls_template(self.picam2.camera_controls)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
I also sometimes just get a black square as image. Restarting the app seems to solve this. Thus something seems to be unstable which is fine, just wanted to report.

Also saving config does not seem to update the last-config json:

```
zev@cam:~/picamera2-WebUI-beta $ cat camera-last-config.json 
{
    "cameras": [
        {
            "Num": 0,
            "Model": "ov5647",
            "Is_Pi_Cam": true,
            "Has_Config": false,
            "Config_Location": "default_ov5647.json"
        }
    ]
}zev@cam:~/picamera2-WebUI-beta $ 
```

monkeymademe, 2025-03-30:
The owned by root I will look into.

The other stuff will be fixed I am working at the moment and have yet to push fixes

closed 2025-04-10 by monkeymademe