monkeymademe/CamUI #122: New camera request

BlueLan0421, 2025-04-14:
Hello, I would like to ask how to use CamUI to control the DFM 37UX226-ML camera. I have already referred to previous examples and added some camera-related information in the camera-module-info.json file. Could you please provide further guidance or examples?

Also, I want to thank you for your previous adjustments regarding exposure time—it worked perfectly!

[image]

[image]

monkeymademe, 2025-04-14:
If you try and use the camera in picamera2 by itself does it work?

The error your getting is due to my script looking for the sensor_modes for the camera but then its failing on picamera2's end... 

I created a diagnostics file - https://github.com/monkeymademe/CamUI/blob/main/diagnostics.py

Run that and give me the output and we can see if there is anything I could do or if this issue should goto the picamera2 fellas

BlueLan0421, 2025-04-15:
Here are the results after running the diagnostics and checking the camera with libcamera.
It seems like the camera cannot be supported with picamera2. ( I guess
But I still want to know if there are any other ways to solve this, or if there are any solutions based on the results from diagnostics.py?
Tks for your kind reply.
[image]

[image]

monkeymademe, 2025-04-15:
I thought about having a switch that ignores the sensor mode if there is an error or its malformed (actually its something I will do for USB webcams that picamera2 on a limited function sorta supports). But I have the feeling that there will be more issues since there is VERY few and weird camera controls coming from picamera2.

I would recommend going to Picamera2's github and submit a request there but then also check out if you can contact libcamera as well. 

I will close this ticket for now but if you get any further / or I manage the sensor mode fix soon we can reopen this

closed 2025-04-15 by monkeymademe

monkeymademe, 2025-04-15:
Actually did one slightly better --- Made this new ticket for the sensor mode if you want to track the progress

https://github.com/monkeymademe/CamUI/issues/123