monkeymademe/CamUI #124: error starting service

galoki, 2025-04-18:
I'm having issues starting CamUI as a service.  Works fine if I run it from the dir (python app.py). 

I cloned using git clone https://github.com/monkeymademe/CamUI.git

My  camui.service is:

[Unit]
Description=CamUI Server
After=network.target
[Service]
Type=simple
ExecStart=/usr/bin/python /home/ronl/CamUI/app.py (this is correct path to app.py)
Restart=always
[Install]
WantedBy=multi-user.target

I get this error starting the service:

File "/home/ronl/CamUI/app.py", line 180, in __init__
Apr 18 09:11:08 pi5cam python[2228]:     self.live_controls = self.initialize_controls_template(self.picam2.camera_controls)
Apr 18 09:11:08 pi5cam python[2228]:                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 18 09:11:08 pi5cam python[2228]:   File "/home/ronl/CamUI/app.py", line 343, in initialize_controls_template
Apr 18 09:11:08 pi5cam python[2228]:     with open("camera_controls_db.json", "r") as f:
Apr 18 09:11:08 pi5cam python[2228]:          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 18 09:11:08 pi5cam python[2228]: FileNotFoundError: [Errno 2] No such file or directory: 'camera_controls_db.json'

any help would be appreciated

monkeymademe, 2025-04-18:
can you run 

`ls /home/ronl/CamUI/app.py`

Give me the output

galoki, 2025-04-18:
ronl@pi5cam:~ $ ls /home/ronl/CamUI/app.py
/home/ronl/CamUI/app.py

I'm assuming you were looking for this?

ronl@pi5cam:~ $ ls -l /home/ronl/CamUI/
total 136
-rwxr-xr-x 1 ronl ronl 70087 Apr 18 09:01 app.py
-rw-r--r-- 1 ronl ronl 14091 Apr 18 09:01 camera_controls_db.json
-rw-r--r-- 1 ronl ronl   216 Apr 18 09:18 camera-last-config.json
-rw-r--r-- 1 ronl ronl  8243 Apr 18 09:01 camera-module-info.json
-rw-r--r-- 1 ronl ronl   399 Apr 18 09:01 connected_cameras_config.json
-rw-r--r-- 1 ronl ronl  1716 Apr 18 09:01 diagnostics.py
-rw-r--r-- 1 ronl ronl  3053 Apr 18 09:01 gpio_map.json
-rw-r--r-- 1 ronl ronl  1071 Apr 18 09:01 LICENSE
-rw-r--r-- 1 ronl ronl  6703 Apr 18 09:01 README.md
drwxr-xr-x 8 ronl ronl  4096 Apr 18 09:01 static
drwxr-xr-x 3 ronl ronl  4096 Apr 18 09:01 templates

monkeymademe, 2025-04-18:
If you pull main branch it will work there was a mistake with the location of the file. Its fixed now

closed 2025-04-18 by monkeymademe

monkeymademe, 2025-04-18:
No worries... was an oversight on my part