monkeymademe/CamUI #125: Logging and error recover is needed when streaming threads crash

hogwell, 2025-04-20:
**Describe the bug**
After a little while, the streaming feed freezes and the app.py needs to be restarted to bring it back.

**To Reproduce**
On a Pi 4 with an IMX477 Hires sensor, run the program in one of the higher resolution modes,
for example, Mode 3 (4056x3040).
After awhile, the video stream will freeze in place and cannot be recovered through the UI.
This problem is less frequent at the lower resolutions.
After the video freezes, running 'top' shows that the auxiliary threads for app.py are all closed, leaving only the original app.py thread still running.

**Expected behavior**
When the CPU cannot keep up with the refresh rate, I would expect the software to catch these errors
in the auxiliary threads and, at a minimum, log and display the error in the UI,
as well as implement some kind of recovery logic.
For example, the exception could be caught and the refresh rate dynamically lowered until the stream is
reliable on the given CPU.

**Desktop (please complete the following information):**
 - Computer: Raspberry Pi 4B
 - OS: Pi OS bookworm
 - Browser Firefox (any)
 - Version 2.0.0 Beta

monkeymademe, 2025-04-25:
Hey sorry for the radio silence. I got a pi4 dug out and ran some tests. Basically as of v2.0.0 just for the video feed the sensor mode adjusted the output resolution and 'encoded' resolution to be the same. This was a mistake on my part. They should be separate for so you can tweak and get the best results for the hardware you are running. For the pi5 its a walk in the park but if you max out the sensor mode (which maxes out the feed resolution) the pi4 starts to fail. Educated guess is if you have a pi4 with more ram the happier it will be (I think mine is 4gb and just does not work on high res.... not even for a little bit)

So there is a couple of solutions right now... 

- Get a pi5... sorry... 
- with v2.0.0 stick with a lower resolution and vow never to change the sensor mode
- download the branch v.2.0.1 (as of this message) and you can tweak the output resolution setting for the video feed (its new) and find the max resolution to sensor mode combo that suits you. There is a new few lines of code that allow for feed recovery, its janky and could fail but if you set to a mode that wont work it just wont but the rest of the system continues to work so you can change the sensor mode again and have it recover... just manually

Otherwise give me some time to iron out a more automatic recovery method... 

My plan to fix this permanently is (this is notes for me):

- Better error reporting / inline alert so you know its gone bad
- If the feed fails have a automated restart and report to the user what happened and offer recommended solution
- Manual restart camera option that does just that.
- When a sensor mode is selected and the feed fails, automatically lower the feed resolution and retry the sensor mode until the feed restarts error free
- If the feed resolutions (all of them) don't work lower the sensor mode and start again from the same resolution as the sensor mode
- Have the system know its on an older pi so highlight there could be an issue in the higher resolution sensor modes and that the above automated recovery takes place.

I feel there will be more...

hogwell, 2025-04-26:
It sounds like you have a good plan for addressing this.
Some kind of logging from other threads would help.
Meanwhile, I think this is the best solution:
> When a sensor mode is selected and the feed fails, automatically lower the feed resolution and retry the sensor mode until the feed restarts error free.