monkeymademe/CamUI #129: Error With Sensor Mode 3, RPi HQ Camera on RPi Zero 2 W

johnson29630, 2025-05-30:
**Describe the bug**
Cannot select 'Sensor Mode 3' for maximum resolution for RPi HQ Camera.

**To Reproduce**
New installation on RPi Zero 2 W running headless. Using SSH/Putty to install and start CamUI with all dependencies. CamUI startes with no noticable issue. 
- Using the web interface on a network computer/browser, switch to 'Camera' view. (camera does work)
- Can switch from Mode 1 to Mode 2 or Mode 3, no issues, camera view displays, can capture images.
- Attempted to select Mode 3 faximum resolution for RPi HQ Camera, camera window does not display live image, will not capture images. The SSH/Putty window shows debug window with 'Error saving profile: [Errno 12] Cannot allocate memory' error
- I can select Mode 0, functionality is restored.

**Expected behavior**
I expect to be able to use Mode 3, as well as Modes 0/1/2

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Browser: Chrome
 - Version 136.0.7103.114 (Official Build) (64-bit)

**Additional context**

RPi Zero 2 W running RPi OS (lite). Followed instructions to install CamUI and dependencies. Specificalu using RPi Zero 2 W for small size and reduced power consumption. 

SSH data captured and attached. Shows all activity once CamUI was started.

Line 1: CamUI started
Line 143: select Mode 1 from web interface (started at Mode 0)
Line 164: select Mode 2 from web interface
Line 185: select Mode 3 from web interface (camera stops working)
Line 200: 'Error saving profile: [Errno 12] Cannot allocate memory'
Line 203: select Mode 0 from web interface (camera function restored)

[CamUI log.txt](https://github.com/user-attachments/files/20525983/CamUI.log.txt)

monkeymademe, 2025-07-03:
yeah there is an issue higher resolution on older hardware... the mjpeg encoder crashes when its too much for the system.

in the branch https://github.com/monkeymademe/CamUI/tree/v2.0.1 I changed the encoder to the jpegencoder. I tested with the pi4 and v3 camera and it seems to be working ok... There is an unknown effect to performance and quality but at least its working for now. 

I don't have a pizero setup for testing so if you could give it ago to see if its better?

Please remember pizero with maxed out resolution might not work so well...

johnson29630, 2025-07-09:
@monkeymademe  I've been traveling, and will test the new version as soon as I am settled in. Thanks for the response. I'm using the Pi Zero 2 W as it's a small package with the camera, and it's low power for battery use. Will update with my results!

monkeymademe, 2025-07-27:
Could you do me a favor and try and run an update... There has been a few updates in picamera2 and libcamera in the last few weeks. Technically should not effect what I am doing but I would like to rule that out

johnson29630, 2025-07-28:
Before my attempt to run CamUI 2.0.1 I had run a system update on July 25 using this command:
`sudo apt update && sudo apt upgrade -y`
Just to be sure I just re-run the system update.
CamUI 2.0.1 still does not launch.

PS: I am not clear on the git command syntax to install specific versions. I used the git web interface to access and download .zip files for the target versions, and manually installed the files to my RPI.

monkeymademe, 2025-08-01:
in the folder for for 2.0.1 type 

`git pull` 

See if it updates the version that might be it.... Maybe